wpilibsuite/GradleRIO #1: Code not deploying

MasterEjay, 2015-02-04:
I'm trying to deploy my code to the RoboRIO using this plugin, but whenever I try to, it always gives me this error. 

http://cl.ly/Zc88

The RoboRIO is connected and I can access it's webpage at `roborio-xxxx.local`

JacisNonsense, 2015-02-04:
Can you run the 'tree' command on your FRC2015/build folder and send me the output? I'll get working on a fix tomorrow when I'm in my working hours. 
Also, which version of GradleRIO are you running?

MasterEjay, 2015-02-04:
On OSX, there is no built in tree command, so I installed it using `brew install tree`. I dont know if this is what you wanted, but heres the output: http://pastebin.com/Y2V5BpCM

MasterEjay, 2015-02-04:
Also, I'm either running 0.1.0 or 0.1.1

JacisNonsense, 2015-02-05:
Hmm, odd. I'll take a look once I get access to our RIO and track down the issue. The latest release shouldn't have any issues as I'm running OSX as well and can still build ^

MasterEjay, 2015-02-05:
Ok, so it looks like it deployed when I updated to the new version. Thanks!

MasterEjay, 2015-02-10:
After I tested it further, it looks like it isn't deploying the code correctly. It says `Successfully deployed`, but it doesn't actually send the code to the RoboRIO.

JacisNonsense, 2015-02-10:
Your src/ files should be moved to src/main/java when using gradle. That should fix the issue

MasterEjay, 2015-02-10:
You mean like merge the gradle `src` with my `src` folder?

JacisNonsense, 2015-02-10:
Move all of your java files from src/ to src/main/java. That should fix the issue if it can't find your code

closed 2015-03-06 by JacisNonsense