wpilibsuite/GradleRIO #6: During configuration phase, gradle fails due to missing WPILIB dependency

lordoku, 2015-11-08:
Per the READEME, I downlaod version 3.0 of the zip file and set up a new project with the robot source. I updated the build file to use our team number and ran the gradlew tasks and got the following errors: 

[image]

At this point Gradle is trying to set up the configuration and has not actually started he build. Any suggestions on how to resolve this issue?

JacisNonsense, 2015-11-08:
Run `gradlew wpi`

lordoku, 2015-11-08:
Same exact error. Like I said, the problem is during the configuration phase, when the build.gradle file is read and the tasks are generated.

[image]

JacisNonsense, 2015-11-08:
Send me a copy of your `build.gradle` file and I'll take a look when I get home

closed 2015-11-08 by JacisNonsense

lordoku, 2015-11-09:
I'm out right now, but I will post it when I get back. Thank you for looking into this issue.

lordoku, 2015-11-14:
Sorry it took so long for me to get back. Here is our build.gradle

    /**
    Template file for classes using GradleRIO
    @author Jaci
    */

    buildscript {
        repositories {
	    mavenCentral()
	    maven { 
                name = "GradleRIO"
                url = "http://dev.imjac.in/maven"
            }
        }
        dependencies {
            classpath group: 'jaci.openrio.gradle', name: 'GradleRIO', version: '+' //Change this line if you wish to Update GradleRIO
        }
    }

    apply plugin: 'java'
    apply plugin: 'idea'
    apply plugin: 'eclipse'
    apply plugin: 'GradleRIO' //Apply the GradleRIO plugin
 
    gradlerio.robotClass = "org.usfirst.frc.team1885.Robot"   //The class for the main Robot Class. Used in manifest
    gradlerio.team = "1885"       //Your FRC team number (e.g. 5333 for team 'Can't C#', or 47 for Chief Delphi)
    //gradlerio.rioIP = "10.53.33.20"                         //Uncomment to specify the IP address of the RIO

    def robotManifest = {
       attributes 'Main-Class': 'edu.wpi.first.wpilibj.RobotBase'
       attributes 'Robot-Class': gradlerio.robotClass
    }

    jar {
        from configurations.compile.collect { it.isDirectory() ? it : zipTree(it) }
        manifest robotManifest
    }

    task genJavadoc(type: Jar, dependsOn: javadoc) {
        classifier = 'javadoc'
        from javadoc.destinationDir
    }

    artifacts {
        archives genJavadoc
    }

lordoku, 2015-11-14:
I saw you made an update and it looks like that fixed our issue. Thank you for looking into this :)

JacisNonsense, 2015-11-14:
No problem :+1:

commit c9c4a99e (2015-11-08): Warn the user if WPI Libraries aren't present to fix build failure during configuration. Fixes #6