wpilibsuite/GradleRIO #11: Official Gradle Plugins Support

333fred, 2016-01-08:
Would you be able to put your plugin up on plugins.gradle.org? Instructions on how to do that are here: https://plugins.gradle.org/docs/submit. This makes it much easier to create a robot project with newer versions of Gradle. Instead of 
```gradle
buildscript {
  repositories {
	mavenCentral()
	maven {
		name = "GradleRIO"
		url = "http://dev.imjac.in/maven"
	}
  }
  dependencies {
    classpath group: 'jaci.openrio.gradle', name: 'GradleRIO', version: '+'			//Change this line if you wish to Update GradleRIO
  }
}
apply plugin: 'GradleRIO'
```
we could simply be able to do:
```gradle
plugins {
    id: 'jaci.openrio.gradle.GradleRIO' version '# here'
}
```

JacisNonsense, 2016-01-08:
I'll see what I can do tonight

oconnelc, 2016-07-16:
Any progress on this? This would be really helpful.

JacisNonsense, 2016-07-16:
I'll see what I can do when the next release rolls around. This may require a bit of changing on GradleRIO's own build system, as well as either uploading to both my personal server and the official plugins server, or adding some redirect trickery to maintain compatibility with existing projects.

HeroCC, 2016-11-23:
Now that the 2016-17 season is coming up, it would be great to have this implemented for the build season; thanks for the great work!

JacisNonsense, 2016-11-27:
I've decided to drop this idea for the time being, or at least until GradleRIO gets a major update. There's a few reasons for this, and I'll list them below:  
- Breaks compatibility. Running GradleRIO on plugins.gradle.org and pulling it from my person dev.imjac.in/maven repository will break every single project that depends on it. I could run this side by side, but managing two publishing targets just to save a few lines of a build.gradle file isn't worth it IMHO.  
- Breaks Toast Projects specifically. GradleRIO is published with two branches, vanilla and toast. The vanilla version is for regular WPILib projects, while the toast version changes some behaviour with WPILib linkages and build deploying to suit Toast projects. On [my maven repo](http://dev.imjac.in/maven/jaci/openrio/gradle/GradleRIO/4.5.0/), this difference is given in the classifier to the project. Unfortunately plugins.gradle.org does not support this, and I don't believe there is a  way to publish a second plugin based on build artifacts (GradleRIO is built twice, changing the value of manifest attribute GradleRIO-Flavour between builds)  
- Moving GradleRIO to plugins.gradle.org means that its dependencies (i.e. NetworkDelegate) will have to be moved, too. This is a fairly minimal effort, but this is just a second instance of the first issue

JacisNonsense, 2016-11-27:
I'll keep this idea incubating for future releases, at which point I may replace the GradleRIO versions on my own repo with a stub that fails a build and prints "we have moved" or something similar

JacisNonsense, 2017-01-11:
This has been included with the new versions. Use plugin `jaci.openrio.gradle.GradleRIO` from now on

closed 2017-01-11 by JacisNonsense