wpilibsuite/GradleRIO #8: Wrong Maven Url

333fred, 2016-01-08:
Please use the release version of the Maven repo, not development. Development is always the latest builds off of master, whereas release is the official version shipped with the plugins. The correct url is http://first.wpi.edu/FRC/roborio/maven/release.

JacisNonsense, 2016-01-08:
On my last check the 'release' maven was missing some artifacts (networktables ARM iirc). Has this been resolved?

333fred, 2016-01-09:
They're all there. The artifacts are copied from our build server to the web-accessible repo, so you might have checked before they were fully finished copying.

333fred, 2016-01-09:
If you do notice something missing, please let me know.

JacisNonsense, 2016-01-09:
Okay, sounds good to me. Will merge :+1:

PR #9 (2016-01-09): Fixed maven url to use the release repository
Fixes #8. Easy enough fix that I can quickly make a pull request for you.
closed 2016-01-09 by JacisNonsense