wpilibsuite/GradleRIO #13: Code not deploying properly?

unon1100, 2016-01-13:
I took a look at https://github.com/Open-RIO/GradleRIO/issues/1 already, that didn't fix my issue.

When I run gradlew deploy, the "Robot Code" area in driver station still remains red, although Communications are still green, and I can connect to my RoboRIO using http://roborio-2239-frc.local

As shown here, gradle says that it deployed successfully. If there is anything that I'm missing, let me know for sure
[image]

unon1100, 2016-01-13:
Here's my tree if you need it - https://gist.github.com/unon1100/5caebba685366683d50c

m3rcuriel, 2016-01-13:
Two questions: 1) If you click "restart robot code" on the Driver Station, does it work? 2) If you ssh into the RoboRIO, is there an FRCUserProgram.jar in the home directory?

unon1100, 2016-01-13:
1) I've tried, no luck
2) I'll check tomorrow morning

JacisNonsense, 2016-01-13:
A few updates were pushed to GradleRIO in the last few days to solve some deployment issues. While connected to the internet, run `gradlew dependencies`. Then you can run `gradlew deploy` again to send the code to your RoboRIO. If the problem persists, please SSH into the roboRIO and send us any error files you come across

unon1100, 2016-01-14:
Nothing in the home directory after deploy. RoboRIO firmware and image is up-to-date.
I ran the deploy with --info, here is the result https://gist.github.com/unon1100/c407893ad5d35bfb48bb

unon1100, 2016-01-14:
Update - I checked a light on the RoboRIO, its communication light is red. The Driver Station said that both ends of communication were working properly, so we're trying switching the RoboRIOs and seeing if it's just a problem with that.

JacisNonsense, 2016-01-15:
Could you send me the output of `ls` under the `home/lvuser` directory of your RoboRIO? If I could have a copy of your `build.gradle`, it would also aid in reproducing the issue. This is an interesting issue indeed

unon1100, 2016-01-15:
I'll get on it tomorrow morning

unon1100, 2016-01-15:
I think I just figured out why. After reimaging and redeploying, the files are there, and the log says that "The directory was not found" when executing the .jar ... pretty sure I just forgot to install Java on the roborio. I'll do that now, haha

unon1100, 2016-01-15:
Looks like that fixed the problem. It'd be good idea to do a check to see if the rio has Java installed when deploying by checking the /usr/local/frc/JRE/bin/java directory to avoid confusion like this in the future. Just a dumb mistake on my part I guess :P

closed 2016-01-15 by unon1100

JacisNonsense, 2016-01-16:
Haha, no problem. I'll add that in :+1: