wpilibsuite/GradleRIO #14: Move to the WPILibVersioning plugin

333fred, 2016-10-16:
We now have a versioning plugin for WPILib, which takes care of figuring out what repository to use. I suggest that you move to using it. You can find the plugin here: https://plugins.gradle.org/plugin/edu.wpi.first.wpilib.versioning.WPILibVersioningPlugin. I don't believe that you can actually apply it to the script yourself, as you can't modify the buildscript after being loaded. But you can apply the plugin to the example script and set up the plugin correctly:

```gradle
WPILibVersion {
    releaseType = official
}
```

This will make sure that you are using the correct repo, and allow others to use the development repo easily if they want to.

JacisNonsense, 2016-10-16:
Hmm, this is neat. I'll take a look into it. There should be a way to add this as a plugin dependency and use it in the WPIProvider class

333fred, 2016-10-16:
I initially tried adding it myself and submitting a pull request, but I wasn't able to get it to work since it's not possible for a plugin to modify the buildscript. You might be able to do something with an uberjar like scenario, but that really goes against the design of gradle itself.

JacisNonsense, 2016-10-16:
Call me out if I'm wrong, but it seems to me that all the plugin is currently doing is adding a maven repository for `http://first.wpi.edu/FRC/roborio/maven` on either the release or dev branches. This is the same as we're doing in GradleRIO [now](https://github.com/Open-RIO/GradleRIO/blob/master/src/main/groovy/jaci/openrio/gradle/WPIProvider.groovy#L24-L41), using the release branch. 

Is there a suggestion that this URL will change at some point in the future?

If I'm not wrong, then perhaps we can add an extension for the branch they want to use, having release as the default, but giving the user the option of dev if they like to live on the wild side

333fred, 2016-10-16:
In terms of what you need to care about, that's correct. It does some other stuff for us by determining what version we should be publishing, but it's not useful for you. As to whether or not the URL will change, there aren't any plans at the moment. I guess you could just add an extension to choose the repo. 

The other thing you'll want to do is make the version of the wpilib dependencies that you add configurable. We're going to publish real versions now, so teams should be able to decide whether they want to take an optional update or not.

JacisNonsense, 2016-10-17:
I'll add the changes to allow for fetching from either the release or development upstreams, as well as an optional version tag, which should be fairly simple, defaulting to a version of `+` (to fetch the latest) if the user doesn't specify the version they want to use. That should fulfil the criteria for this issue

closed 2016-11-27 by JacisNonsense

JacisNonsense, 2016-11-27:
See README.md ([diff](https://github.com/Open-RIO/GradleRIO/commit/1e3b813990f48d756c9c0b0458107dfecfe365e6)) for details

333fred, 2016-11-29:
@JacisNonsense, and important note for your README: there are beta releases on the release channel. A better set of names might be public/internal, but I didn't want to break everyone since we'd chosen release/development before. `beta`, `rc`. and regular releases will be published to the release channel.

JacisNonsense, 2016-11-30:
@333fred I've updated the README to reflect this important differentiation

commit 79ab1f62 (2016-11-27): Add ability to specify WPI version/branch. Close #14