wpilibsuite/GradleRIO #15: Joystick.getRawButton() with IDs higher than 8 return the value of button 8

19lmyers, 2016-10-26:
Tested this with identical code, one a standard WPI project and the other a GradleRIO project. Querying getRawButton(int id) with an ID higher than 8 returned the status of button 8 in the GradleRIO project.

JacisNonsense, 2016-10-26:
GradleRIO doesn't change the inner workings of WPILib or any of your code for that matter, there is no way this is a GradleRIO issue.
What else can you tell me about your GradleRIO environment?

19lmyers, 2016-10-26:
I downloaded the latest release .zip, extracted it, and set up with IntelliJ IDEA. Here's my repository: https://github.com/Team980/Baseball-Shooter-Gradle

I have a similar repository without GradleRIO that does not exhibit this problem: 
https://github.com/Team980/Baseball-Shooter-Java

I can't think of anything irregular that might be the cause.

JacisNonsense, 2016-10-26:
Can you see what version of WPILib is running in either case? You can see this in the dependencies section of eclipse or IDEA

When you say an ID higher than 8, how many physical buttons are on your controller? What does the WPILib raw project return in such a case where the ID is higher than 8?

JacisNonsense, 2016-10-26:
The maven repository is the official repository of WPILib (See #14), and `wpilibJavaFinal-0.1.0-SNAPSHOT.jar` is the latest, stable published version on the release tree. 

You can get the version eclipse is running by looking at the RobotBase class source, and somewhere around the 226th line, you'll see a file write containing the WPI version. You can also confirm this on the Driver Station. 

The version GradleRIO is fetching from the Maven repository is `2016 Java Beta5.0`

19lmyers, 2016-10-26:
Eclipse (WPI Project) doesn't list a version. It's attached as
wpilib-sources.jar

The GradleRIO project gives it a version: wpilibJavaFinal-0.1.0-SNAPSHOT.jar

I noticed that version was coming from a maven repository. Could the
repository be outdated?

There are at least 11, possibly 12 numbered buttons on the joystick.

The WPI project returns the proper buttons (9 when 9 is queried, etc).

19lmyers, 2016-10-27:
My GradleRIO project is showing 2017.1.0-beta-1

My WPI project has 2016 Release 5

JacisNonsense, 2016-10-27:
Huh, interestingly enough this is an issue with WPILib itself, or rather their publishing to maven

To try and offer a quick explanation, the maven tree that WPILib is published to ([here](http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilibj/wpilibJavaFinal/) for WPILibJ) doesn't follow a proper versioning system, causing gradle to not be able to identify and invalid cache of the library. You'll notice only one version is listed, `0.1.0-SNAPSHOT`, however within that version there appears to be subversions for each build (look at the dates column). Each of these 'subversions' have a different release version. Compare this to other maven repositories ([GradleRIO's for reference](http://dev.imjac.in/maven/jaci/openrio/gradle/GradleRIO/)), you can see the individual versions are listed properly.

The importance of this is that when only one version is listed, gradle will say "yup, I'm up to date" and won't bother fetching new artifacts (bar specifying `--refresh-dependencies`). The library version loaded on my system is all the way back from `2016 Beta 5`, however yours appears to be the latest. This makes sense given that my Gradle caches haven't been cleaned since late 2015. 

WPILib's eclipse plugins package the library with the plugin itself, which is why the versions differ. Since you're running a beta version for 2017, some anomalies are to be expected. This should be _partially_ resolved by #14 when I put the changes in, however that will depend on how WPILib's maven functions in the future. 

I will submit an issue on WPILib's main repository and see if I can get some changes made. Hold tight.

JacisNonsense, 2016-10-30:
This should be fixed now that WPILib's release branch has been setup properly for 2016 releases

JacisNonsense, 2016-10-30:
Run `./gradlew build --refresh-dependencies` to refetch Gradle's dependency cache. This should fix most of if not all compilation and runtime issues.

closed 2016-10-30 by JacisNonsense

19lmyers, 2016-10-30:
Can confirm that running said command downloaded the 2016 version of WPILib. However, the External Libraries directory in the IDEA project tree does not display the newly pulled versions. (A newer version of NetworkTables was also downloaded.)

Will test with robot during the week.