wpilibsuite/GradleRIO #16: GradleRIO and third-party libraries

19lmyers, 2017-01-07:
Starting with the 2017 update to WPILib, classes such as CANTalon are no longer provided as a part of the library. In addition, the only way I've found to restore functionality such as CANTalon usage is to install a CTR library via a EXE installer. I am reasonably confident that this method is not directly compatible with GradleRIO. 

How could I use these third-party libraries with GradleRIO?

JacisNonsense, 2017-01-08:
By taking a quick look at the CTRE libraries, they are available in two forms, as an exe installer, and as a zip archive. It is currently possible to use both of these to install 3rd party libs to GradleRIO. Until I push an update out which will make this more streamlined, you can use the following methods to link the 3rd party libs.

On the [CTRE Tools Website](http://www.ctr-electronics.com/hro.html#product_tabs_technical_resources), choose either the `CTRE Toolsuite Installer`, or the `CTRE Toolsuite (no installer) package`. Either will do, but for now I would recommend the installer one.

Add the following line to your `build.gradle`'s `dependencies { }` block. 
```gradle
compile fileTree(dir: "<tooldir>", include: '**/*.jar')
```
Depending on which binary you downloaded, you will substitute `<tooldir>` with a different string.

 If you downloaded the installer, replace `<tooldir>` with `${new File(System.getProperty("user.home"), "wpilib/user/java/lib").absolutePath}`

If you downloaded the non-installer package, replace `<tooldir>` with `<extraction_path>/java/lib`, where `<extraction_path>` is where you extracted the zip archive to.

This will compile against the CTRE Jar for your robot program. If you're using stock-standard GradleRIO, this will build into your final .jar. If you're using Toast, it will require a little bit of extra setup to include it in your binary (since toast doesn't build fat jars). I will discuss this in a followup comment.

The problem now is that CTRE actually doesn't link their `.so` file (their native code for the roborio) into the .jar, so you'll have to install that, too. Add the following lines to the bottom of your `build.gradle`, it's best if you don't ask how this part works. Substitute `<tooldir>` with whatever you substituted it with before.

```gradle
fileTree(dir: "<tooldir>", include: '**/*.so').each { File f ->
    gradlerio.deployers += [ to: "/usr/local/frc/lib/${f.name}.jar", from: f.file ]
}
```

JacisNonsense, 2017-01-08:
If you ARE using TOAST, you will need to follow all the steps above with some extra setup:
In your `dependencies {}` block, add the following line, substituting `<tooldir>` for whatever you substituted it with above.
```gradle
toastLibrary fileTree(dir: "<tooldir>", include: '**/*.jar')
```

JacisNonsense, 2017-01-08:
As a guide, your `build.gradle` should look a little something like [this](https://gist.github.com/JacisNonsense/bfe30eb366ad3d679e88fc19e9652c25)

JacisNonsense, 2017-01-11:
Closed for #17

closed 2017-01-11 by JacisNonsense