wpilibsuite/GradleRIO #22: Unable to deploy

bob80333, 2017-01-15:
After updating my team's roboRio to the 2017 firmware and updating our router, I tried to deploy and got this error message:

`Failed SFTP PUT: C:\Users\Jared\IdeaProjects\SammyRoboRio\build\libs\SammyRoboRio.jar -> Remote21:/home/lvuser`

[Stacktrace](https://hastebin.com/izilusoqug.vbs)
[build.gradle](https://hastebin.com/gihefupisi.http)

19lmyers, 2017-01-15:
The same EXACT thing is happening to me, except it's called Remote3.

bob80333, 2017-01-15:
The number kept incrementing as I kept trying to deploy

JacisNonsense, 2017-01-15:
You need to run `./gradlew build` before `./gradlew deploy`, i.e. `./gradlew build deploy`. This was a change made during the 2016-2017 migration to decouple building and deploying

bob80333, 2017-01-15:
OK, thanks

closed 2017-01-15 by bob80333