wpilibsuite/GradleRIO #25: Talon SRX path blocked by firewalls

kyle-github, 2017-01-17:
I'm a mentor at a US high school.  The site that the Gradle script tries to copy the Talon SRX files from is in India (based on the URL).  That site is blocked by the school's firewall (as are a large number of other sites).  Are there any other locations that we can the Maven-ized version of this?  And, how would we set that up?  We are Gradle noobs.

FRC 2017.   Using IntelliJ on Windows, Mac and Linux.  Or at least trying to...

JacisNonsense, 2017-01-17:
The website is imjac.in, my personal web server. The server itself is based
in Sydney, Australia, however the domain name is not. There isn't a way for
me to put this somewhere else, as my server has to run a nightly script to
update CTRE's stuff (they don't publish it on maven). I'm afraid there
isn't much I can do on my end

19lmyers, 2017-01-17:
Why won't CTRE bother to publish it on Maven?

kyle-github, 2017-01-17:
Rats.

We are trying to use the "native" example shown in the README file, but without any luck as well.   We were able to find the CTRE site and libraries and copied them locally to try to avoid pulling them from the Internet.   The dependencies now look like:

`dependencies {
    compile wpilib()
    // compile talonSrx()
    compile fileTree(dir: 'libs', include: '**/*.jar')
    native  fileTree(dir: 'libs', include: '**/*.so')
}`

However, we get an error on the "native" part:

Error:(17, 0) Cause: startup failed:
build file '/home/pce/IdeaProjects/TestPlatform/build.gradle': 17: unexpected token: dir @ line 17, column 22.
       native  fileTree(dir: 'libs', include: '**/*.so')
                        ^

1 error

If I comment out the "native" part, it does copy the .jar file and I can push that to the robot.   But then it does not run because it does not have the .so :-(

kyle-github, 2017-01-17:
Ugh, the formatting did not work there at all.  Let me try again.

dependencies {
    compile wpilib()
    // compile talonSrx()
    compile fileTree(dir: 'libs', include: '**/*.jar')
    native  fileTree(dir: 'libs', include: '**/*.so')
}

JacisNonsense, 2017-01-17:
Whoops, native is a reserved word in java, I didn't realise that got shared
with gradle DSL. I'll push a fix shortly

JacisNonsense, 2017-01-17:
As for why CTRE don't publish to maven, I'm not sure. @ozrien any details?

JacisNonsense, 2017-01-17:
@kyle-github use GradleRIO `2017.1.1` and use `nativeLib` instead of `native`.

kyle-github, 2017-01-17:
Great, I will try this tomorrow (it is late night where I am).  Thanks for the work on this.   IntelliJ is much happier due to you!

ozrien, 2017-01-17:
"As for why CTRE don't publish to maven, I'm not sure. " I don't really know what this means.

  About half of teams use Java, and the majority of those teams seem to use eclipse. If that changes we will adapt accordingly. 

 Our nonwindows installer should suffice for any one who wants to automate or create a package installer.

JacisNonsense, 2017-01-17:
@ozrien Maven is insanely useful for artifact publishing. Not only is it used by almost every java build system, but it's even used by [WPILib to publish their releases](http://first.wpi.edu/FRC/roborio/maven/release). Having WPILib itself published to maven, but having what is probably the number 1 most used device library (talonSrx) not published to maven means we need to have alternate solutions, such as the [server scripts I run](https://github.com/Open-RIO/GradleRIO/blob/master/server_script/ctre.rb) to make sure the community has access to these libraries from a maven interface. Given that the overwhelming majority of the community uses Java OR C++ to develop their robot code, wouldn't it make sense to make these libraries easily accessible for everyone, regardless of their development environment preferences, without having to download and install manually (something that can cause version or environment conflicts for large teams of programmers)?

kyle-github, 2017-01-18:
@ozrien: I really appreciate the efforts your company goes through to make these available, but having a Maven-ready source for the JARs and native libraries would be much more "standard."   To build on Jaci's point, even with other build systems Maven repositories are used for binary objects.  It is one of the most common ways to store that sort of data.

Our team is using IntelliJ now due to the constant problems with performance and stability with Eclipse.  And, as a member of the greater Java world (I work for Oracle), it is clear that Eclipse is no longer the industry darling.

I definitely understand that you want to be able to track your customers.  Would you consider setting up a Maven repo that required a CTRE login to access?

The FRC development environment has come a long way in the last five years from almost purely proprietary and difficult to manage to almost completely open.   Supporting Maven and Groovy (thanks, Jaci!) helps even more.

floogulinc, 2017-01-18:
@ozrien Even if you don't start using Maven, could you at least open source the library on GitHub or similar?

msoucy, 2017-02-17:
The lack of a Maven package is definitely problematic for teams that want to verify their code via travis-ci or similar. Currently, my team is using the injac.in maven repository just to be able to test - having a canonical maven repository would be greatly preferred

closed 2017-08-19 by JacisNonsense