wpilibsuite/GradleRIO #26: Deploy fails with no internet connection?

snekiam, 2017-01-21:
When I have no internet at a competition, and I need to make a small change, will I still be able to deploy?  I have watched gradlew build fail more than once with no internet, but once I connect and run the same command, it works fine.  I can then disconnect again, and builds seem to work for a little while, then fail again.  I'm not sure what exactly is going on here, but it seems like maybe the gradle cache is expiring or something.  I can still browse to .gradle\caches\modules-2\files-2.1, and all the libs seem to be present when a build fails.

JacisNonsense, 2017-01-21:
Gradle is looking if any libraries are out of date. Run `./gradlew build deploy --offline` if you don't have an internet connection and it will use the local version

closed 2017-01-21 by snekiam

JacisNonsense, 2017-01-21:
All good, I'll make a note of it in the readme