wpilibsuite/GradleRIO #31: Add javadoc for Talon SRX library

floogulinc, 2017-02-02:
There is a javadoc provided in the Talon library zip, could we could include that so it is available from gradle?

Also, it may be better to call it something like CTRLib, instead of talonSRX in the gradle file since it is actually more general than just the Talon SRX.

JacisNonsense, 2017-02-02:
I'll take a look at JavaDoc publishing. 

I think talonSrx is just fine (most people only use it for talon srx), however I can add an alias

floogulinc, 2017-02-02:
Yeah, the name is fine. Thanks for looking into it.

firecrafty, 2017-10-09:
Since CTRE has since released Pheonix which breaks backward compatibility and adds wayyyy more stuff than just the CANTalon libraries, I think a renaming of the dependency, even if it is an alias, is absolutely necessary.

JacisNonsense, 2017-10-09:
I should mention this has already been done in the queued 2018 refresh of GradleRIO
https://github.com/Open-RIO/GradleRIO/blob/refresh2018/src/main/groovy/jaci/openrio/gradle/wpi/WPIDependencies.groovy#L85-L88

firecrafty, 2017-10-09:
Awesome!

JacisNonsense, 2017-12-17:
This is now in the vendor's hands with https://github.com/CrossTheRoadElec/Phoenix-frc-lib/pull/5

JacisNonsense, 2018-01-05:
OpenRIO's mirror of the CTRE library now includes the sources as an artifact along with the phoenix release.

closed 2018-01-05 by JacisNonsense