wpilibsuite/GradleRIO #33: update CTRLib?

cyocom, 2017-02-21:
Currently it seems like http://dev.imjac.in/maven/ only has CTRLib v 4.4.1.9, there has been a 4.4.1.10 and a 4.4.1.11 out since then, can we get those added to maven repo?

Thanks for all you do!

JacisNonsense, 2017-02-22:
Normally the server scripts [here](https://github.com/Open-RIO/GradleRIO/tree/master/server_script) would update with new releases. This was, however, under the assumption that the final part of the version number would flow over the the second last part when it becomes saturated (e.g. the release after 4.4.1.9 would be 4.4.2.0). As this isn't the case, alphabetical sorting doesn't work (the server script takes the 'latest' one at the time of checking, which in this case is incorrectly identified as 4.4.1.9.)

This has been fixed. 4.4.1.11 was pushed to maven but 4.4.1.10 was skipped

closed 2017-02-22 by JacisNonsense

cyocom, 2017-02-27:
Thanks again!

We're loving GradleRIO this year compared to doing the normal WPILib method, it really help us when everyone can use their own IDE and you don't have to worry about your class paths being synced up. (gradle does it for you, woo!)

GradleRIO has allowed us to have quick turnaround times, creating a more efficient programming workflow and has developed an environment where, for the first time in a long time, "deploying code" is no longer a statement that comes with the fear of minutes of lost time on the robot.

JacisNonsense, 2017-02-27:
I'm glad to hear it! It's good to see GradleRIO is helping people out c: