wpilibsuite/GradleRIO #34: Long Deploy Times

snekiam, 2017-03-02:
We've been experiencing excruciatingly long deploy times using GradleRio this whole season (sometimes 40+ seconds, usually around 30).  This has been happening on multiple laptops and multiple roborios, so I don't think the problem resides there (but it might).  It is also happening with offline deploys, so it isn't a network speed problem.  I'll have a whole log for you tonight (~ 2 hours from now), but in the meantime, I have two questions:
1) Is this normal?  I thought I remember reading somewhere that deploys were supposed to be ~15 seconds.
2) Is there anything we can do to shave time off?

Our build.gradle is embedded below.

```
plugins {
    id "java"
    id "eclipse"
    id "idea"
    id "jaci.openrio.gradle.GradleRIO" version "2017.1.2"
}

frc {
    team = "3322"
    robotClass = "org.usfirst.frc.team3322.Robot"
    rioIP = "10.33.22.2"
}

dependencies {
    compile wpilib()
    compile talonSrx()
    compile files('lib/navx_frc.jar')
}

def robotManifest = {
  attributes 'Main-Class': 'edu.wpi.first.wpilibj.RobotBase'
  attributes 'Robot-Class': frc.robotClass
}

jar {
    from configurations.compile.collect { it.isDirectory() ? it : zipTree(it) }
    manifest robotManifest
}

task wrapper(type: Wrapper) {
    gradleVersion = '3.2.1'
}

snekiam, 2017-03-03:
Log from our deploy in intellij:
[image]

```7:08:39 PM: Executing external tasks 'build deploy --offline'...
:compileJava
:processResources UP-TO-DATE
:classes
:jar
:assemble
:compileTestJava UP-TO-DATE
:processTestResources UP-TO-DATE
:testClasses UP-TO-DATE
:test UP-TO-DATE
:check UP-TO-DATE
:resolveNativeDeps
:build
:determine_rio_address
============ FINDING ROBORIO ============
-> mDNS (roboRIO-3322-frc.local)...
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
-> USB (172.22.11.2)...
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
-> Static IP (10.33.22.2)...
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
--> SUCCESS
============ ROBORIO FOUND ============
:deploy
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.

BUILD SUCCESSFUL

floogulinc, 2017-03-03:
This is consistent with our experience on my team. You may want to also look at the logs from the roboRIO as you deploy. On ours it looks like it's restarting the robot code multiple times.

JacisNonsense, 2017-03-03:
I wouldn't call these deploy times 'excruciating'. 

Part of the reason it is taking so long is that your RoboRIO can't be found by mDNS. This is an issue with your network, and you should see similar results on the eclipse deployer. For now, set `frc.ip = 10.33.22.2` in your build.gradle and you should see speeds increase by about 6 seconds.

The other part is user libraries. GradleRIO, like WPILib's Eclipse Java Plugins, will deploy user libraries each deploy. In the C++ version of GradleRIO, they are checked against a cache, but in Java, they are not, as native libraries are usually quite sparse. I'll look into caching this in the future

snekiam, 2017-03-03:
We do have the navx and talonsrx libraries included in our deploy, which would increase our deploy times.  But I already had that static IP set in my build.gradle, and I even tried setting it in the hosts file - it didn't seem to make much of a difference.  Would it be possible to make a task that doesn't deploy libraries (as they are presumably already on the robot?) for quick code changes?  Or is everything packaged into the one jar file.
The deploy log above took 1 minute and 30 seconds, a far cry from the "10-15 seconds" quoted in the readme.  But I do agree, excruciating probably isn't the right word.

On another note, thanks for the awesome work you're doing on this project Jaci.  Other than the deploy times being longer than I remembered in eclipse (I may be remembering wrong), I love what this plugin has opened up for our team.

JacisNonsense, 2017-03-03:
Try setting `frc.rioHost = 10.33.22.2` instead, which bypasses mDNS.
If you want to build but not deploy, you can run `gradlew build`. Deploy includes the libraries as to reduce the amount of ssh sessions spawned.

I agree that 1m 30s are a bit ridiculous, but I can't say I can reproduce that. The worst I've seen is 45 seconds from a clean build. I'll work on the caching tonight and see what we can do to get those times down.

snekiam, 2017-03-03:
If you need me to provide any logs, I am more than happy to.  I'll have some time with the robot tomorrow, Saturday and Sunday, and I'll see if I can deploy some very very basic code (no external libs, etc) and see how long it takes.  I can also give you the output of some stacktraces if it'll help.

Tommsy64, 2017-03-03:
Having a deploy task that doesn't deploy libraries for quick code changes would be immensely useful.  We get deploy times around 50 seconds-1 minute.  Lowering that would save much time in the normal development workflow where the time spent waiting for code to upload adds up after hundreds of small deploys.

(Thanks for your work!)

JacisNonsense, 2017-03-03:
@snekiam Not the competition robot I hope :P

@Tommsy64 Currently, in GradleRIO-C, we check the library against a checksum stored on the RoboRIO since the last deploy. This makes sure that libraries are only deployed if they need changing. This functions the way you'd expect from a deploy-without-libraries-task, except for cases where the RoboRIO has been reimaged or new libraries have been added. This is what I'm going to implement, since I believe it should cut deploy times down immensely

JacisNonsense, 2017-03-03:
@snekiam @Tommsy64 @floogulinc Plugin version `2017.1.4` has been published and includes the new checksum code. It hasn't been tested since I don't have a robot available here but it should work just fine. Please let me know the results.

Tommsy64, 2017-03-03:
That was fast! Will the changes in version `2017.1.4`  be reflected in the repository? Is commit [8fb5f4c](https://github.com/Open-RIO/GradleRIO/commit/8fb5f4caaa90062ba53606e74d857632e89ec153) version `2017.1.3`?

I will report results as soon as I have access to a robot. (Should be within 24 hours)

JacisNonsense, 2017-03-03:
@Tommsy64 I'm on flaky campus WiFi right now (the ruckus of week 1, not a DHCP lease to be found), so the git push was a little delayed. For this same reason, `2017.1.3` was an attempted push of `2017.1.4` that didn't finish. `2017.1.2` was the previous stable.

floogulinc, 2017-03-03:
@JacisNonsense I will test when I have a chance. Thanks. (I wish I had a roboRIO for testing here)

snekiam, 2017-03-03:
Unfortunately, this doesn't seem to make a huge difference for us.  It seems like the checks for the libs are failing every time. This is one (of many like it) prints happening during deploy:
`RoboRIO Library libwpilibJavaJNI.so.debug.MD5.MD5.MD5.MD5.MD5 out of date! Updating Library`

snekiam, 2017-03-03:
Also another interesting point: On each deploy, another .MD5 is added to the name of each library.

JacisNonsense, 2017-03-04:
@snekiam oh dear. Try `2017.1.5`, I believe this should solve that issue

snekiam, 2017-03-04:
Awesome work Jaci!  The same deploy that took 1+ minute before is now under 20 seconds.  Thank you so much!

closed 2017-03-04 by snekiam

JacisNonsense, 2017-03-04:
@snekiam Awesome, glad I could help