wpilibsuite/GradleRIO #35: nativeZip behavior changed

WawerOS, 2017-03-10:
nativeZip complains that it can't resolve  cscore-1.0.2-athena-uberzip.zip, we used to be able to placate it with a copy in libs/ and using --offline but it now complains again.

WawerOS, 2017-03-10:
I have confirmed using wget and curl that GET works for the repository.

JacisNonsense, 2017-03-11:
You shouldn't be placing any WPI core library in your own libs/ file. GradleRIO manages this [for you](https://github.com/Open-RIO/GradleRIO/blob/master/src/main/groovy/jaci/openrio/gradle/wpi/WPIPlugin.groovy#L59-L64)

WawerOS, 2017-03-11:
Could that be the problem? Cause gradlerio also stalls getting the appropriate library, all others download fine.

JacisNonsense, 2017-03-11:
Does the same issue occur if you remove your cscore loaded in libs/, and set `wpi.cscoreVersion="1.0.1"` ?

JacisNonsense, 2017-03-11:
Also a proper log would be useful

closed 2017-08-19 by JacisNonsense