wpilibsuite/GradleRIO #36: InputStream is closed

donovanMaas, 2017-04-22:
Gradle is constantly unable to deploy to the RoboRIO. I am building before deploying.

`Failed SFTP PUT: /Users/64008508/GitHub/Titanium/build/libs/Titanium.jar -> Remote3:/home/lvuser`

`1 of 2 session (50%) failed`

`Session #2 failed: java.lang.RuntimeException: java.lang.RuntimeException: org.hidetake.groovy.ssh.operation.SftpFailureException: Failed SFTP PUT: /Users/64008508/GitHub/Titanium/build/libs/Titanium.jar -> Remote3:/home/lvuser: (SSH_FX_FAILURE: An error occurred): java.io.IOException: inputstream is closed`

[stacktrace](https://pastebin.com/16v2MzWk)

JacisNonsense, 2017-04-26:
I vaguely remember having this issue before, and fixing it. I'll take a closer look when I'm back in Perth.

JacisNonsense, 2017-08-19:
This issue should be resolved in the last few releases

closed 2017-08-19 by JacisNonsense