wpilibsuite/GradleRIO #38: resolveNativeDeps tries to download nonexistent file

NoahGleason, 2017-08-16:
Our code just started failing build (I first noticed it last night) because resolveNativeDeps attempts to get opencv-jni-linux-arm.jar from [http://first.wpi.edu/FRC/roborio/maven/release/org/opencv/opencv-jni/3.2.0/opencv-jni-3.2.0-linux-arm.jar](http://first.wpi.edu/FRC/roborio/maven/release/org/opencv/opencv-jni/3.2.0/opencv-jni-3.2.0-linux-arm.jar), which 404s. I'm pretty sure this is just WPI deleting a file and isn't an issue on your end, but if WPI deleted that file intentionally, the dependency needs to be changed. I'm using version 2017.1.5.

Tommsy64, 2017-08-16:
It looks like your `opencvVersion` is set to `+`. This will default to the latest version. On August 14, version 3.2.0 was released which no longer contains the `linux-arm` artifact.

In `build.gradle`, set the wpi.opencvVersion back to `3.1.0` to use the previous version.

```
wpi {
    wpilibVersion = "+"
    ntcoreVersion = "+"
    opencvVersion = "3.1.0"
    cscoreVersion = "+"
}

dependencies {
...
```
Note: *The `wpi` block has to be above the `dependencies` block.*

NoahGleason, 2017-08-17:
That worked, thanks! Maybe temporarily alias opencvVersion="+" to 3.1.0 instead of 3.2.0 so other people don't get the same nasty shock I did?

JacisNonsense, 2017-08-17:
This issue is because WPI changed their naming convention a few days ago, with the introduction of their new build pipeline. Everything that was linux-arm is now linuxathena, and some projects (like the C++ headers) have been restructured. I need to set aside some time to changing GradleRIO to facilitate these changes. I'm aiming for this weekend, but there are other deadlines I must meet first.

For now, run openCV 3.1.0 and run gradle with the `--offline` flag after your first build. If anyone else is running into the same issue, type "bump" in a comment on this thread so you can be notified when it is fixed. It should be fixed within a week or two.

JacisNonsense, 2017-08-19:
This issue has been fixed with GradleRIO release `2017.2.3`. Note that in this release of GradleRIO (and future releases), only OpenCV >=3.2.0 will be supported due to the naming convention change on WPI's end. If you wish to downgrade OpenCV below 3.2.0, you will have to use GradleRIO `2017.1.5`.

Sorry for the huge version number jump, Gradle's publishing service was giving some issues so I had to skip some version numbers.

commit 12478cb2 (2017-08-19): Add build exception for the new OpenCV naming convention for athena. From this version of GradleRIO onwards, only openCV > 3.2.0 will be supported. Fixed #38
closed 2017-08-19 by JacisNonsense