wpilibsuite/GradleRIO #39: OpenCV Native libraries could not be loaded

firecrafty, 2017-09-07:
I am using 2017.2.3. I have redeployed and reimaged.

```
OpenCV Native Libraries could not be loaded.
Please try redeploying, or reimage your roboRIO and try again.
java.lang.UnsatisfiedLinkError: no opencv_java320 in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1857)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1119)
	at edu.wpi.first.wpilibj.RobotBase.initializeHardwareConfiguration(RobotBase.java:181)
	at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:194)
********** Robot program starting **********
Error at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:249): ERROR Unhandled exception: java.lang.NullPointerException at [org.usfirst.frc.team1018.lib.util.MultipleJoystickButton.<init>(MultipleJoystickButton.java:15), org.usfirst.frc.team1018.robot.OI.<init>(OI.java:24), org.usfirst.frc.team1018.robot.OI.getInstance(OI.java:51), org.usfirst.frc.team1018.robot.Robot.robotInit(Robot.java:47), edu.wpi.first.wpilibj.IterativeRobot.startCompetition(IterativeRobot.java:64), edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:247)]
WARNING: Robots don't quit!
---> The startCompetition() method (or methods called by it) should have handled the exception above.
NT: server: client CONNECTED: 10.10.18.20 port 56381
```

JacisNonsense, 2017-09-08:
Looks like I'm missing a path. I'll take a deeper look soon

Note to self: check `/usr/local/frc/lib`, opencv_java320 might be deploying to the linux subdirectory

JacisNonsense, 2017-09-08:
Looks like WPI changed where libopencv_<vers>.so is stored inside the jar. Use OpenCV 3.1.0 and GradleRIO 2017.1.5 for now, since there's a new version of GradleRIO queued that will be rewritten, so I'll change this behavior while I'm at it

firecrafty, 2017-09-08:
Just to clarify, I have to declare the OpenCV version explicitly in my build.gradle, correct?

JacisNonsense, 2017-09-08:
Yes, in your build.gradle:
```
wpi {
    opencvVersion = "3.1.0"
}
```

NoahGleason, 2017-09-17:
replying to be notified when this gets fixed

JacisNonsense, 2017-09-18:
Noah, is the above 3.1.0 fix working, or is there another fix that is required to be more urgent?

NoahGleason, 2017-09-18:
Using 1.3.0 is working great, I just want to know when I can switch.

JacisNonsense, 2017-09-29:
I'm going to bump real quick for when I take a look at this issue in the newer releases.

It may be possible to make 3.1.0 and 3.2.0 both work in the same plugin version, given some clever version checking. For <=3.1.0, the artifact is opencv-jni-VERS-linux-arm, while from 3.2.0 onwards it is opencv-jni-VERS-linuxathena

JacisNonsense, 2017-10-30:
This should be fixed now. 2017.10.30 for 2017, 2018-beta-2c for 2018 beta

tkdberger, 2017-10-30:
Hmm. Possibly unrelated, but while using `2017.10.30` attempting to set exposure on a USB camera with `setExposureAuto()` is throwing a `VisionException`. This wasn't happening before today, which makes me think there's tickery afoot on Gradle's end. I've tried manually setting `opencvVersion` to  `"3.1.0"` in the wpi block (right after the `deploy` one.)

firecrafty, 2017-10-31:
Would the OpenCV version affect that? If I'm not mistaken, USBCamera is part of cscore which is part of wpilib.

JacisNonsense, 2017-10-31:
I believe that's a cscore thing. Send me your stacktrace (gradlew clean build --stacktrace) and I'll see where this issue lays

tkdberger, 2017-10-31:
Clean build doesn't spit out any stacktrace, just the normal Gradle log (command: `./gradlew clean build --stacktrace`):

```
Picked up _JAVA_OPTIONS: -Dawt.useSystemAAFontSettings=on -Dswing.aatext=true -Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel
Starting a Gradle Daemon, 1 stopped Daemon could not be reused, use --status for details
:clean
:compileJava
:processResources NO-SOURCE
:classes
:jar
:assemble
:compileTestJava NO-SOURCE
:processTestResources NO-SOURCE
:testClasses UP-TO-DATE
:test NO-SOURCE
:check UP-TO-DATE
:build

BUILD SUCCESSFUL in 16s
3 actionable tasks: 3 executed
```

I don't have the robot with me at this exact moment, but I can get the trace from the actual code crash when I do (should been soon:tm:.)

tkdberger, 2017-11-02:
[Here's the trace from the DS.](https://pastebin.com/a4QrTJ7u) I'll see if there's a more detailed version available in the RIOlog (RIO log?)

[In case it's relevant, here's our `Robot.java` as well.](https://github.com/BHSSFRC/3494_2017_repo/blob/master/3494_2017_repo/src/org/usfirst/frc/team3494/robot/Robot.java) This version is from before we started with GradleRIO, but there hasn't been any changes to the cameras in code since then (just what debug info goes to the SmartDashboard I believe.)

JacisNonsense, 2017-12-17:
Since there's been a lot of recent updates, OpenCV bindings should be fixed now, at least in the beta branch. Feel free to reopen if it's still failing

closed 2017-12-17 by JacisNonsense