wpilibsuite/GradleRIO #40: Support for directory-based structure

andriyasdf, 2017-09-09:
GradleRIO should drop file-based project structure support when generating a new project and instead use directory-based. This would make it easier for version control to manage, as files are split up into smaller segments.

JacisNonsense, 2017-09-10:
Can you give me a better idea of what you're suggesting? Are you talking about Gradle project, Eclipse project, IntelliJ projects, or something else? Examples would be appreciated

andriyasdf, 2017-09-10:
Whoops, I meant to specify that I’m talking about IntelliJ projects. When running the command “gradlew idea,” it would be ideal if this generates a directory-based project instead. Can GradleRIO support this format?

andriyasdf, 2017-09-10:
Now looking at the IDEA plugin for Gradle, I'm assuming you don't have much control over this process. I guess we'll have to wait and see what gradle decides to do.

closed 2017-09-10 by andriyasdf

JacisNonsense, 2017-09-10:
You can still make that kind of thing happen with multi-project builds. Just include the 'idea' plugin in all your projects, and run `./gradlew idea` from the root project. That'll generate an idea project that includes an idea module for each subproject. We did it in 2016 [here](https://github.com/FRC5333/2016-Champs)