wpilibsuite/GradleRIO #41: Trouble unzipping Toolsuite-Java-4.4.1.14.jar

stewsters, 2017-09-15:
I am having some trouble with running this in 
`
$ ./gradlew build
Starting a Gradle Daemon (subsequent builds will be faster)
:compileJava UP-TO-DATE
:processResources UP-TO-DATE
:classes UP-TO-DATE
:jar FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':jar'.
> Could not expand ZIP '/home/bloodred/.gradle/caches/modules-2/files-2.1/thirdparty.frc.ctre/Toolsuite-Java/4.4.1.14/94160dbe0ab528f71d83ae75c900f1190984377f/Toolsuite-Java-4.4.1.14.jar'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 5.309 secs
`

I tried cleaning and removing the file so it would download again, and it happened again.  I tried unzipping the file from the CLI and I got this error:

`
~/code/robotics/rioTest$ unzip /home/bloodred/.gradle/caches/modules-2/files-2.1/thirdparty.frc.ctre/Toolsuite-Java/4.4.1.14/94160dbe0ab528f71d83ae75c900f1190984377f/Toolsuite-Java-4.4.1.14.jar
Archive:  /home/bloodred/.gradle/caches/modules-2/files-2.1/thirdparty.frc.ctre/Toolsuite-Java/4.4.1.14/94160dbe0ab528f71d83ae75c900f1190984377f/Toolsuite-Java-4.4.1.14.jar
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of /home/bloodred/.gradle/caches/modules-2/files-2.1/thirdparty.frc.ctre/Toolsuite-Java/4.4.1.14/94160dbe0ab528f71d83ae75c900f1190984377f/Toolsuite-Java-4.4.1.14.jar or
        /home/bloodred/.gradle/caches/modules-2/files-2.1/thirdparty.frc.ctre/Toolsuite-Java/4.4.1.14/94160dbe0ab528f71d83ae75c900f1190984377f/Toolsuite-Java-4.4.1.14.jar.zip, and cannot find /home/bloodred/.gradle/caches/modules-2/files-2.1/thirdparty.frc.ctre/Toolsuite-Java/4.4.1.14/94160dbe0ab528f71d83ae75c900f1190984377f/Toolsuite-Java-4.4.1.14.jar.ZIP, period.
`

Can you check to see if you can rm that file and re-download it?  It might have gotten corrupted or something.

JacisNonsense, 2017-09-16:
I'll check tonight, I'm out of the house right now. I'll see if I can find
the issue

JacisNonsense, 2017-09-17:
Oddly enough the Zip doesn't appear to be corrupted on my servers' end. I've successfully unzipped it on both linux and windows. Run `./gradlew build --refresh-dependencies`. If that still fails, run again with `wpi { talonSrxVersion = "4.4.1.11" }` in your build.gradle and report back

stewsters, 2017-09-17:
Refreshing the dependencies worked.  Closing issue.

closed 2017-09-17 by stewsters