wpilibsuite/GradleRIO #42: Can't specify NavX library version

NoahGleason, 2017-09-17:
According to https://github.com/kauailabs/navxmxp/blob/master/ReleaseNotes.md, the most recent version of the NavX library is 3.0.316, but Gradle can't find the library when I specify navxVersion = "3.0.316". If this can't be fixed, I'd like to know what version "+" represents so I can hard code that value to avoid unpleasant surprise updates.

JacisNonsense, 2017-09-18:
The release notes Kuaui have put up are wrong. The latest version is 3.0.331 (fetched here http://dev.imjac.in/maven/thirdparty/frc/kauai/Navx-Java/) 

+ Will default to the latest version in that above url. Running `gradlew dependencies` will also show you local dependency versions.

JacisNonsense, 2017-09-18:
Sorry, turns out `+` defaults to a dotpoint in markdown. Above comment should read as such

closed 2017-09-18 by JacisNonsense