wpilibsuite/GradleRIO #43: Revert CTRE Version

jhh, 2017-09-19:
It looks like GradleRIO is updating the TalonSRX dependency to an alpha version (5.0.2.1) that has code-breaking [API changes](https://www.ctr-electronics.com/downloads/release_notes/ALPHA_RELEASE_NOTES). Is there anyway to specify that I want to use version 4.4.1.14 of the CTRE jar?

Thanks!
P.S. This project is a great contribution to the community - very much appreciated!

jhh, 2017-09-19:
OK, I *knew* there was already a way to specify the version (there in the README ;-).

closed 2017-09-19 by jhh

JacisNonsense, 2017-09-20:
Glad you found it :)

I'm working on some infrastructure stuff to allow users to put 'versions: recommended-2017' or something similar that will fetch recommended, known stable versions. This'll likely be a callback to my server