wpilibsuite/GradleRIO #45: GradleRIO not deploying to robot

19lmyers, 2017-10-04:
For some reason GradleRIO is no longer deploying code to our robot. Our code deploys fine when copied to a fresh Eclipse/WPI project, but all deploys with GradleRIO result in the driver station displaying "No Robot Code".

Gradle deploy log (I ran ```gradle build``` online previously to this):
```
7:37:22 PM: Executing external task 'deploy'...
:determine_rio_address
============ FINDING ROBORIO ============
-> mDNS (roborio-980-frc.frc-robot.local)...
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
--> SUCCESS
============ ROBORIO FOUND ============
:deploy
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
Host key checking is off. It may be vulnerable to man-in-the-middle attacks.
RoboRIO Library libHALAthena.so out of date! Updating Library
RoboRIO Library libHALAthena.so.debug out of date! Updating Library
RoboRIO Library libntcore.so out of date! Updating Library
RoboRIO Library libntcore.so.debug out of date! Updating Library
RoboRIO Library libCTRLibDriver.so out of date! Updating Library
RoboRIO Library libCTRLibDriver.so.debug out of date! Updating Library
Remote2#87|ldconfig: libraries libfrccanfirmwareupdate.so and libfrccanfirmwareupdate.so.orig in directory /usr/local/frc/lib have same soname but different type.
Remote2#87|ldconfig: /usr/local/frc/lib/libfrccanfirmwareupdate.so is not a symbolic link
Remote2#87|

BUILD SUCCESSFUL

Total time: 28.203 secs
7:37:50 PM: External task execution finished 'deploy'.
```

I don't think the library error above is the problem as it also showed up in the Eclipse log.

build.gradle:
```
plugins {
    id "java"
    id "eclipse"
    id "idea"
    id "jaci.openrio.gradle.GradleRIO" version "2017.2.3"
}

frc {
    team = "980"
    robotClass = "com.team980.robot2017.Robot"

    rioHost = "roborio-980-frc.frc-robot.local"
    rioIP = "10.9.80.2"
}

wpi {
    wpilibVersion = "2017.3.1" //Provided by Eclipse
    talonSrxVersion = "4.4.1.14" //Provided by CTRE Toolsuite
}

dependencies {
    compile wpilib()
    compile talonSrx()

}

def robotManifest = {
  attributes 'Main-Class': 'edu.wpi.first.wpilibj.RobotBase'
  attributes 'Robot-Class': frc.robotClass
}

jar {
    from configurations.compile.collect { it.isDirectory() ? it : zipTree(it) }
    manifest robotManifest
}

task wrapper(type: Wrapper) {
    gradleVersion = '3.2.1'
}
```

edelmanjm, 2017-10-04:
Does the log on the driver station say anything interesting?

19lmyers, 2017-10-04:
From the looks of the log, the code isn't even running when we deploy via
GradleRIO.

edelmanjm, 2017-10-04:
Can you paste it here?

firecrafty, 2017-10-04:
Yes, please paste it. It very well could be deploying, but errors when it runs resulting in no robot code.

JacisNonsense, 2017-10-04:
Downgrade to GradleRIO 2017.1.5 and set opencvVersion to 3.1.0. This is a known issue due to some WPILib changes, and will be fixed before the next GradleRIO release, but this is a temporary workaround.

JacisNonsense, 2017-10-30:
This should be fixed now. 2017.10.30 for 2017, 2018-beta-2c for 2018 beta

closed 2017-10-30 by JacisNonsense