wpilibsuite/GradleRIO #46: Don't default to dynamic dependency versions

bherbst, 2017-10-16:
Defaulting to "+" for all dependency versions is dangerous, as it can pull in new versions that cause breaking changes.

Two great examples of this are #45 and #43 , both of which caused builds to fail due to breaking changes in the dependencies. If teams are using GradleRIO with the default configuration of not specifying the versions themselves, they will be completely unaware that the version is changing and will simply see a failed build.

*Ideally* teams would be forced to declare the dependencies themselves, but I definitely appreciate the convenience of bundling everything together under methods like `wpilib()`.

A compromise might be to add a version to the convenience methods, such as `wpilib("2017.3.1")`, and split out the different components more (e.g. `openCv("3.1.0")`).  This would make the versioning explicit and keep all dependency information in the dependencies closure instead of putting some of it in the WPI plugin configuration. Doing so would also make it more obvious what dependencies are pulling in, when some like open CV are currently more or less hidden unless the team knows where in the GradleRIO repository to dig around.

If the goal is to encourage keeping those libraries up to date, we could output an info/warning message as appropriate when new versions become available. As a bonus, that wouldn't require any server-side code to report out of date information as was mentioned in #43.

I'd be happy to submit a PR if we can agree on an approach.

JacisNonsense, 2017-10-16:
This is already on the chopping block for the newer version of GradleRIO (see refresh2018 branch).

Essentially, default versions are hardcoded in the plugin, but the plugin will make a request to a URL containing the 'recommended' versions for libraries for that version of the plugin (i.e. 2017 releases would work with 2017 stable libraries). Users can still override this if they choose, but it automatically applies to all GradleRIO-managed libraries if not otherwise set. If the URL can't be reached, GradleRIO will fallback to one of: a local cache of that URL, or if not found, the hardcoded values.

GradleRIO is going under a massive rework for 2018, including better extensibility and target support (through [EmbeddedTools](https://github.com/JacisNonsense/EmbeddedTools/tree/extensionModel)), as well as better dependency management. You can keep up to the progress by watching the refresh2018 branch as it changes. The rework will also combine the Java and C++ (native) versions of GradleRIO into a single plugin, and allow for targets other than the roborio.

We try to wrap library support under a single method since WPILib's internal linking structure changes drastically from year to year, and there are some special caveats that have to be taken into account when  adding as a dependency to a project. Although components like NTCore and CSCore are separate libraries, they are still necessary for WPILib to function, so to me it doesn't make sense to not wrap this all under one method.

As a final note, the reason we initially used `+` was to automatically update if new releases came out during the season. Usually, WPILib updates for beta should be under a separate maven repository all together (development instead of release), but it seems this year that it's a bit of a mishmash.

bherbst, 2017-10-16:
Sounds great, thanks!

I haven't started poking around the 2018 branch too much yet, but I'll do some more poking.

closed 2017-10-16 by bherbst

JacisNonsense, 2017-10-16:
Awesome :)
If you want to keep up to date, watch this space for issues tagged `help wanted`. Once I migrate GradleRIO to the newer model of EmbeddedTools there should be some issues that the community can dive on. The 2018 branch is currently based on EmbeddedTools in the model space, but for some practical reasons EmbeddedTools has been moved to the extension space (see the extensionModel branch), and also contains more of the linking and library logic.