wpilibsuite/GradleRIO #47: Unable to find target RIO when able to ping every other way

Redrield, 2017-11-02:
When testing to see if it was my computer that was having troubles connecting to the robot for purposes of code deploy, I wanted to see if GradleRIO would be able to determine the RIO address. I copied the quickstart buildscript into an empty directory structure and ran `gradle discoverRoborio`. The command came back saying that it was unable to discover the target, and was aborting because of a config choice.

I was able to find the RIO's mDNS in avahi-browse -alr. I was able to ping the robot's IP (in this case 10.40.69.2), and I was able to see content dispatched from its web dashboard (Telling me to get silverlight in this case). So the problem must lie with GradleRIO.

JacisNonsense, 2017-11-02:
Please post a log with `--debug` on your network. Theoretically the IP address should hit it just fine. Can you also SSH into the rio?

Redrield, 2017-11-02:
I was able to ssh into the RIO when gradlerio wasn't able to find it. I'll get you the debug log next time I have some time with our robot.

ddaymn, 2017-11-07:
I had the same issue.  I had a password set for the admin user.  When I cleared it I could deploy

JacisNonsense, 2017-11-07:
Again, I can't fix anything if I don't have any logs available. Please post them so this issue can get fixed ASAP.

As for the admin password, I believe that's a documented problem with the RoboRIO. The admin password should be blank for FRC use.

SPCOxion, 2017-11-07:
Seems like the `Unable to find target roborio` error is a red herring.

When I ran `gradlew deploy`, it gave me the "Unable to find target" error.
However, when I ran `gradlew deploy -w`, it gave me a different error. The log indicates that it was using the correct IP as well.

After adding the IP to `build.gradle`, both logs remain the same.

I've uploaded both logs to [this](https://gist.github.com/SPCOxion/98cf6fc3b0494e87e76df87c21224056) gist. Hopefully this helps a bit.

JacisNonsense, 2017-11-07:
Please run with `--debug`, it's impossible for me to tell the root otherwise.

Redrield, 2017-11-07:
After much finnicking, I finally got GradleRIO to at least discover my RIO. Not sure about deployment but that'll be a different issue

To anyone on Linux having issues, I think the thing I did that solved it was making sure that avahi-daemon was started. That's the last thing I did before it started seeing my RIO. Just making sure that it's started (and optionally enabled) in systemd could well solve your issues.

Any Winblows or crapOS users, I have no advice

JacisNonsense, 2017-11-08:
I'm keeping this open since it's got a few more people involved in the issue. I'll see if I can do some more hunting to find the issue on Windows, Mac seems unaffected

JacisNonsense, 2017-11-09:
This issue should now be solved via EmbeddedTools v2017.11.09 ([commit here](https://github.com/JacisNonsense/EmbeddedTools/commit/589aab876b9fce591a59194eb7372ad27cdff44f))

For 2017: Use GradleRIO 2017.11.09
For 2018 (beta only): Use GradleRIO 2018-beta-2d

closed 2017-11-09 by JacisNonsense