wpilibsuite/GradleRIO #49: Guidance on adding Log4j

clebio, 2017-11-05:
Hi, I've been using this project successfully for building, deploying, and watching `riolog`. I'm not at all familiar with Gradle though, but would like to use log4j for logging on the roborio. Would you be willing to give me some pointers on how to add external dependencies? The basic stumbling block I'm having is just that the log4j JARs aren't getting copied to the Rio on deploy. This is probably super straight-forward, but I'm slow like molasses reading and writing Gradle, right now.

JacisNonsense, 2017-11-06:
Hey there,
You can add extra dependencies to your project with the `dependencies` block. In the case of log4j, it would look something like this (from the [log4j website](https://logging.apache.org/log4j/2.x/maven-artifacts.html)):
```gradle
dependencies {
  compile group: 'org.apache.logging.log4j', name: 'log4j-api', version: '2.9.1'
  compile group: 'org.apache.logging.log4j', name: 'log4j-core', version: '2.9.1'
}
```

GradleRIO will package this together with your main code when it gets sent off to your RoboRIO. If you're using an IDE like Eclipse or IntelliJ, run the ide task (`./gradlew <your ide>`) to refresh the workspace to include the dependencies

clebio, 2017-11-06:
So, when I tried that, it gives me an error as though it couldn't find log4j (though maybe this error is something else; I'm not sure):
```
Error at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:249):
ERROR Unhandled exception: java.lang.NoClassDefFoundError: java/beans/PropertyChangeEvent at [org.apache.logging.log4j.core.LoggerContext.updateLoggers(LoggerContext.java:654), org.apache.logging.log4j.core.LoggerContext.updateLoggers(LoggerContext.java:641), org.apache.logging.log4j.core.LoggerContext.setConfiguration(LoggerContext.java:547), org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:617), org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:634), org.
apache.logging.log4j.core.LoggerContext.start(LoggerContext.java:229), org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:242), org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:45), org.apache.logging.log4j.LogManager.getContext(LogManager.java:174), org.apache.logging.log4j.LogManager.getLogger(LogManager.java:648), org.usfirst.frc1775.TigerBytes.commands.GyroLogger.<clinit>(GyroLogger.java:16), org.usfirst.frc1775.TigerBytes
.Robot.robotInit(Robot.java:22), edu.wpi.first.wpilibj.IterativeRobot.startCompetition(IterativeRobot.java:64), edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:247)]
```

JacisNonsense, 2017-11-06:
Log4j is being loaded correctly, but it's missing a class that _should_ be a part of the RoboRIO JRE (the installed java). 

Are you using 2017 or 2018-beta? I know 2018-beta includes java beans classes, and from a class list snapshot from 2 years ago (2016), the java beans classes are missing (this should be the same class-set on the 2017 deployment). 

I believe Log4j will be functional for 2018, but it appears as though the 2017 JRE is missing the required classes to run it.

clebio, 2017-11-06:
The plot thickens! :) Is the `master` branch 2017? I see there's a `2018-beta`. If I'm on master, should I try the 2018-beta branch -- would that possibly just fix it?

JacisNonsense, 2017-11-06:
Master is indeed 2017. 2018-beta is for beta testing teams only, the code won't run if you don't have access to the 2018 beta image.

clebio, 2017-11-07:
Ok, that makes sense. I will bide my time until 2018 GA. :) Thanks for clarify things!

closed 2017-11-07 by JacisNonsense