wpilibsuite/GradleRIO #52: Unable to build static libarary

Wishbonea, 2017-11-24:
When compiling a static library, static archiving fails with the following error on Windows:

```
C:\Users\akemp\.gradle\gradlerio\toolchains\bin\arm-frc-linux-gnueabi-ar.exe: unrecognized option `--sysroot'
Usage: C:\Users\akemp\.gradle\gradlerio\toolchains\bin\arm-frc-linux-gnueabi-ar.exe [emulation options] [-]{dmpqrstx}[abcDfilMNoPsSTuvV] [--plugin <name>] [member-name] [count] archive-file file...
       C:\Users\akemp\.gradle\gradlerio\toolchains\bin\arm-frc-linux-gnueabi-ar.exe -M [<mri-script]
 commands:
  d            - delete file(s) from the archive
  m[ab]        - move file(s) in the archive
  p            - print file(s) found in the archive
  q[f]         - quick append file(s) to the archive
  r[ab][f][u]  - replace existing or insert new file(s) into the archive
  s            - act as ranlib
  t            - display contents of archive
  x[o]         - extract file(s) from the archive
 command specific modifiers:
  [a]          - put file(s) after [member-name]
  [b]          - put file(s) before [member-name] (same as [i])
  [D]          - use zero for timestamps and uids/gids
  [U]          - use actual timestamps and uids/gids (default)
  [N]          - use instance [count] of name
  [f]          - truncate inserted file names
  [P]          - use full path names when matching
  [o]          - preserve original dates
  [u]          - only replace files that are newer than current archive contents
 generic modifiers:
  [c]          - do not warn if the library had to be created
  [s]          - create an archive index (cf. ranlib)
  [S]          - do not build a symbol table
  [T]          - make a thin archive
  [v]          - be verbose
  [V]          - display the version number
  @<file>      - read options from <file>
  --target=BFDNAME - specify the target object format as BFDNAME
 optional:
  --plugin <p> - load the specified plugin
 emulation options:
  No emulation specific options
C:\Users\akemp\.gradle\gradlerio\toolchains\bin\arm-frc-linux-gnueabi-ar.exe: supported targets: elf32-littlearm elf32-bigarm elf32-little elf32-big plugin srec symbolsrec verilog tekhex binary ihex

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':createFrcUserProgramStaticLibrary'.
> A build operation failed.
      Static library archiver failed while archiving libfrcUserProgram.a.
  See the complete log at: file:///C:/Users/akemp/Downloads/cpp/build/tmp/createFrcUserProgramStaticLibrary/output.txt

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

* Get more help at https://help.gradle.org

BUILD FAILED in 11s
4 actionable tasks: 1 executed, 3 up-to-date
```

Bug can be reproduced by changing NativeExecutableSpec to NativeLibrarySpec in quickstart cpp project and building on Windows.

For some reason, compilation only fails on Windows, it appears to work fine on Linux.

JacisNonsense, 2017-11-24:
I've merged the PR and will update the plugin version tomorrow (after cherrypicking the commits to 2018-beta also). If you need anything built urgently, you can use `binaries.all` and access the args for the tools on the toolchain manually.

JacisNonsense, 2017-11-26:
Fixed in `2017.11.27` and `2018-beta-3a`

closed 2017-11-26 by JacisNonsense