wpilibsuite/GradleRIO #54: Migration guide for earlier 2017 versions?

19lmyers, 2017-12-02:
My robot project is on 2017.1.5 and I'd like to migrate it to the new 2017.11.27 version in preparation for 2018, but the configuration layout of build.gradle appears to be completely different such that a simple version adjustment won't suffice. 
Importantly, I'm not sure how to set parameters such as rioIP / port / host in the new syntax, which seems to take up a lot more space than the old syntax.

JacisNonsense, 2017-12-03:
You have to adopt the new structure shown in the quickstart, since GradleRIO got a full rewrite to better accommodate future updates, as well as supporting coprocessors. 

In order to set your RoboRIO IP/Host, you set them by adding to the 'addresses' list on the RoboRIO target. In the README, (in Full Spec), it directs you [here](https://github.com/JacisNonsense/EmbeddedTools#spec)

e.g.
```gradle
target("roborio", jaci.openrio.gradle.frc.RoboRIO) {
            team = TEAM
            addresses << 'my_custom_host_or_ip'
 }
```

All these properties are defined in the EmbeddedTools repository.

JacisNonsense, 2018-06-24:
Since we're well past 2017, I'm going to close this now

closed 2018-06-24 by JacisNonsense