wpilibsuite/GradleRIO #58: Simulation + HAL Extension Support

JacisNonsense, 2017-12-17:


jwhite66, 2018-04-07:
You working on this?  Or leaving it here for others to pick up?  :-).

JacisNonsense, 2018-04-07:
It's in a vaguely workable state in src: https://github.com/Open-RIO/GradleRIO/tree/master/src/main/groovy/jaci/openrio/gradle/sim

Only thing is Gradle won't let a process run in the daemon process and die when you CTRL+C (it'll just live in the background until you kill it manually, consuming that daemon). It could be solved by using the same thing we do for debug builds (generate a .sh/.bat file during `./gradlew build` and notify the user to run it), but that kicks the user into another thing they have to do, which is suboptimal. Now that I think about it, we could probably have that as part of the build process instead of dedicating a task to it.

If you want to work on it I'd be more than happy to merge. Most of the logic is in there, but it was far from stable because of the above, which is why I disabled it for the 2018 stable build here: https://github.com/Open-RIO/GradleRIO/blob/master/src/main/groovy/jaci/openrio/gradle/sim/JavaSimulationTask.groovy#L67-L69

If you're testing, just keep in mind the published version of halsim lowfi is broken on windows specifically for some reason, I've been meaning to look into it.

JacisNonsense, 2018-06-24:
Closed since this is mostly implemented and just awaiting some 2019 changes in the root wpilib repo. Its new home is in #115

closed 2018-06-24 by JacisNonsense