wpilibsuite/GradleRIO #61: Cannot version pin for CTRE library versions lower than 5.0

rafibaum, 2018-01-05:
I've been working on getting our local projects updated to the latest GradleRIO. I was trying to version pin our project to the old TalonSRX libraries until they complete the new Phoenix framework but ran into some trouble when specifying the library versions in the build.gradle. Here's the exact error I got:
`Could not find openrio.mirror.third.ctre:CTRE-phoenix-java:4.4.1.14.
Searched in the following locations:
    http://first.wpi.edu/FRC/roborio/maven/release/openrio/mirror/third/ctre/CTRE-phoenix-java/4.4.1.14/CTRE-phoenix-java-4.4.1.14.pom
    http://first.wpi.edu/FRC/roborio/maven/release/openrio/mirror/third/ctre/CTRE-phoenix-java/4.4.1.14/CTRE-phoenix-java-4.4.1.14.jar
    http://dev.imjac.in/maven/openrio/mirror/third/ctre/CTRE-phoenix-java/4.4.1.14/CTRE-phoenix-java-4.4.1.14.pom
    http://dev.imjac.in/maven/openrio/mirror/third/ctre/CTRE-phoenix-java/4.4.1.14/CTRE-phoenix-java-4.4.1.14.jar
    https://raw.githubusercontent.com/Open-RIO/Maven-Mirror/master/m2/openrio/mirror/third/ctre/CTRE-phoenix-java/4.4.1.14/CTRE-phoenix-java-4.4.1.14.pom
    https://raw.githubusercontent.com/Open-RIO/Maven-Mirror/master/m2/openrio/mirror/third/ctre/CTRE-phoenix-java/4.4.1.14/CTRE-phoenix-java-4.4.1.14.jar`

The only repositories out of these that the CTRE libraries are in is dev.imjac.in and the Maven-Mirror repository. The GitHub based Maven-Mirror seems to only have the newest CTRE libraries (5.0+) so trying to use older versions with that repository is pointless. The dev.imjac.in repository does have the older versions of the CANTalon library, but GradleRIO seems to be using the wrong repository path to find them. While GradleRIO searches in `http://dev.imjac.in/maven/openrio/mirror/third/ctre/CTRE-phoenix-java/` the correct location is `http://dev.imjac.in/maven/thirdparty/frc/ctre/Toolsuite-Java/`.

I think there's two possible solutions here:

1. Move all the old versions of the libraries over to the new GitHub-based Maven mirror. It's probably easier for @JacisNonsense to manage just one Maven mirror as opposed to two.
2. Make it so GradleRIO searches in the correct location for the CTRE libraries in `dev.imjac.in`.

rafibaum, 2018-01-05:
I think I understand what the issue is. The package Jaci has set for the CTRE libraries is `openrio.mirror.ctre.CTRE-phoenix-java` and that's just being appended to all the repositories. I'm going to open up a PR with the Maven mirror to add in an old version of the CTRE libraries in the appropriate format.

JacisNonsense, 2018-01-05:
The old CTRE Toolsuite libraries (< 5.0) are not compatible with the 2018 Control System due to some breaking changes on CTRE's side. For this reason, I've chosen to keep the old 2017 toolsuite on the old maven mirror. Since CTRE Toolsuite <5.0 isn't compatible with 2018, you aren't able to use it with GradleRIO 2018 (or any build system for 2018, for that matter)

closed 2018-01-05 by JacisNonsense