wpilibsuite/GradleRIO #62: Incompatibility with Eclipse Plugins

JacisNonsense, 2018-01-06:
Due to the single admin SSH session, the frcUserProgram file conflicts with that being deployed by the eclipse plugins, causing a potential deadlock in the deploy process since frcKillRobot.sh doesn't affect other users' owned processes

calcmogul, 2018-01-10:
This deadlock behavior occurs even on a freshly reimaged roboRIO with the Eclipse deploy never being run.

JacisNonsense, 2018-01-10:
I can't seem to recreate this issue. I'll try a reimage

calcmogul, 2018-01-28:
I can't reproduce this anymore with the v16 image and GradleRIO 2018.01.22.

JacisNonsense, 2018-01-28:
Huh, alright then. I'll keep this open but label it as such. I've tried multiple times but I can't seem to reproduce it either.

Anyone: bump this thread if you see an issue like this

closed 2018-06-24 by JacisNonsense