wpilibsuite/GradleRIO #63: Add Javadocs for commonly-used libraries

aswerdlow935, 2018-01-06:
I'm not sure if this is something that can be done from gradle, but adding the javadocs along with the existing source files would be super helpful. (for intellij at least) I can manually add the javadoc, but with constant library updates and several computers, adding the javadoc would be super useful

JacisNonsense, 2018-01-06:
The JavaDocs are already made available.

IntelliJ will prefer to use provided sources over javadocs, so if you run `./gradlew idea`, it will link both the compiled jar and the sources jar. If you inspect a library class, you can still view the comments of the class.

[image]

aswerdlow935, 2018-01-06:
Ha yes I saw that...I was talking about the *javadoc* javadoc that opens in a browser...but then again maybe I'm the only person who uses the open external documentation feature instead of built-in hower to preview. For some reason Intellij doesn't want to generate it and needs a javadoc link/file

JacisNonsense, 2018-01-14:
I think this is a close. If you want to see the javadoc in a browser, all vendors appear to publish their javadocs online that you can view.

closed 2018-01-14 by JacisNonsense