wpilibsuite/GradleRIO #64: Running GradleRIO open Command launches additional Command Prompt

JDaniel29, 2018-01-07:
Whenever I run `gradlew shuffleboard` or `gradlew smartdashboard`, all that occurs is that my project is built and an additional command prompt is opened to the project directory. The weird thing is that no errors are thrown, and I can open the jar files of the tools by double-clicking on them or running the `java -jar` commands. However, the programs don't open when I run the gradle commands.

JacisNonsense, 2018-01-07:
They are intended to spawn a new command window, but that command window is also intended to run the shuffleboard jar. Can you send a screenshot of what you're seeing?

JDaniel29, 2018-01-08:
Here you go:

The CMD that pops up:
[image]

The output in the terminal:
[image]

JacisNonsense, 2018-01-17:
I noticed this same issue when the `java` command was not on the system PATH. Can you add your JDK/JRE bin directory to your system path?

JDaniel29, 2018-01-21:
I added the bin for my JDK and JRE to the path and it still is only throwing up the blank command prompt. However, this time the command prompt has a link to the bat file in the header.

[image]

JacisNonsense, 2018-01-21:
Try running `./gradlew shuffleboard -Pheadless` and run the file it tells you to run

JDaniel29, 2018-01-21:
OK, when I ran this command and the file, the shuffleboard launched.

firecrafty, 2018-08-31:
Is there any fix planned for this? This is really inconvenient and its easier to just download and keep the jar on my desktop.

JacisNonsense, 2018-08-31:
See #127

closed 2018-09-23 by JacisNonsense