wpilibsuite/GradleRIO #67: RIO Firmware check sould be added.

iboyperson, 2018-01-10:
When building a project with GradleRIO the deploy claimed success but when run with the official ant build scripts the deploy errored because our RIO has not had the firmware updated. A similar feature would be a good Idea to have in GradleRIO as the build was actually not successful.

JacisNonsense, 2018-01-12:
I've added this to the 2019 Roadmap, although I may implement it before the end of the season

iboyperson, 2018-01-12:
Great Thanks!

david476, 2018-01-13:
My team is working on setting up OpenRIO and we weren't able to deploy until we flashed the RoboRIO for this season. Would it at least make sense to add a warning in the docs/ readme?

JacisNonsense, 2018-01-14:
A lack of updated firmware doesn't cause a deploy to fail. It will still send over the code, but the code won't run since it will crash immediately. If your RoboRIO just straight up doesn't have an image on it, of course the deploy is going to fail.

JacisNonsense, 2018-07-01:
This is implemented now, forgot to close earlier

closed 2018-07-01 by JacisNonsense