wpilibsuite/GradleRIO #70: Intellij IDEA Gradle Refresh runs indefinitely

robot-rover, 2018-01-12:
Hello,
I am having issues using this plugin with Intellij IDEA. Whenever I attempt to do a gradle refresh, the process gets stuck on `Gradle: Resolve dependencies of "roboRIO:native_cscore_athena`. Because the operation never finishes, IDEA never updates the dependency list and when I add dependencies to my build.gradle. For example, I can add `LoggerFactory.getLogger()` to my code and compile fine, but Intellij won't add typing completion and tells me the classes are not found/missing.
Any ideas about how I could fix this?

JacisNonsense, 2018-01-12:
I think @HeroCC was having this same issue. My recommendation is to not link your gradle project with IntelliJ, but instead run `./gradlew idea` out of the IDE and running your builds from the command-line. This appears to be an IDE compatibility thing.

I also found this which may be of use: 
https://stackoverflow.com/questions/38209851/intellij-idea-takes-a-long-time-when-refreshing-a-gradle-project/42766110#42766110

flybotix, 2018-01-12:
We just figured out a workaround last night.  The exact same thing happened to us this week in Eclipse while trying to import our repo from scratch.  We figured out that in our case it was related to inter-project gradle dependencies (i.e. ```compile project(:common)```).  That may not be the root cause, but the symptom is exactly what is described above.  I also don't know if we'll re-encounter this issue later in the season, but the workaround should still work worst-case.

Here is the workaround.
Say "e0570d3" is the hash of the git repo that has WPILIB 2017 and is just before the commit which points to WPILIB 2018.
```git checkout e0570d3
[import eclipse gradle project]
git checkout master
Refresh each of the eclipse projects```

We have a bit more testing to do (such as adding/removing dependencies and then running ```./gradlew eclipse``` again).

flybotix, 2018-01-13:
Quick update - after the workaround of importing into Eclipse with 2017 WPI, the problem re-manifests when closing and re-opening Eclipse.  We're re-structuring the gradle setup now so it is a single gradle project.  More testing to come.

robot-rover, 2018-01-13:
Not importing the project as a gradle project and running builds from the command line has alleviated the problem, but it is somewhat inconvenient. Is this something that it would be possible to fix in gradleRIO?

flybotix, 2018-01-14:
We collapsed our multi-gradle project into a single gradle project in the same repo and no one has had a single issue.  We now have to do code reviews specifically to make sure none of the display-related items are called from the robot (e.g. colors, etc).  It's manageable.

JacisNonsense, 2018-01-14:
@robot-rover the IDE maintainers are the ones to bug about this problem, gradle doesn't dictate how these plugins function. As a rule of thumb, I don't import the gradle project and let the command line do the work since these IDE plugins have been buggy since the beginning of time itself.

JacisNonsense, 2018-01-14:
I can try and do something about it but I can't guarantee anything given the nature of the problem

HeroCC, 2018-01-14:
Talked with Jaci about this issue and for any other teams having the same issue a patch should be out in a few hours.

JacisNonsense, 2018-01-15:
Thanks to @HeroCC for finding the root cause of the issue, turns out IntelliJ is doing something weird with classpath configurations causing them to not resolve, causing a deadlock. This is fixed in 2018.01.15

commit 34b602a1 (2018-01-15): Fix #70
closed 2018-01-15 by JacisNonsense