wpilibsuite/GradleRIO #73: Gradle dependency versions don't follow specified dependency versions

19lmyers, 2018-01-20:
I attempted to override the Shuffleboard and WPILib versions used by the plugin, but my version overrides are being ignored by Gradle (as evidenced by the output of ```gradlew dependencies```).
Notably, ```gradlew wpi``` does output the proper overridden versions.
```
wpi {
    wpilibVersion = "2018.2.2"
    shuffleboardVersion = "1.1.1"
}
```
Configuring the wpi block as specified caused the plugin to build with the desired WPILib version, but executing ```gradlew shuffleboard``` still incorrectly runs 1.0.0.

closed 2018-01-21 by JacisNonsense

19lmyers, 2018-02-14:
This is happening again with the new WPILib update (2018.3.1). ```gradlew wpi``` still outputs my properly overridden version, but now the FRC Driver Station still lists the old version, which means the new one isn't being used for compilation.

JacisNonsense, 2018-02-14:
Run `./gradlew dependencies` and send me the output, I can't recreate this issue with GradleRIO 2018.01.22. It's possible the Driver Station is reporting the version wrong

19lmyers, 2018-02-14:
```
 [Version 10.0.17093.1000]
(c) 2017 Microsoft Corporation. All rights reserved.

C:\Users\19lmy\IdeaProjects\2018>gradlew dependencies

> Task :dependencies 

------------------------------------------------------------
Root project
------------------------------------------------------------

apiElements - API elements for main. (n)
No dependencies

archives - Configuration for archive artifacts.
No dependencies

compile - Dependencies for source set 'main' (deprecated, use 'implementation ' instead).
+--- edu.wpi.first.wpilibj:wpilibj-java:2018.2.2
+--- edu.wpi.first.ntcore:ntcore-java:4.0.0
+--- edu.wpi.first.wpiutil:wpiutil-java:3.0.0
+--- org.opencv:opencv-java:3.2.0
+--- edu.wpi.first.cscore:cscore-java:1.1.0
+--- openrio.mirror.third.ctre:CTRE-phoenix-java:5.2.1.1
\--- openrio.powerup:MatchData:2018.01.07

compileClasspath - Compile classpath for source set 'main'.
+--- edu.wpi.first.wpilibj:wpilibj-java:2018.2.2
+--- edu.wpi.first.ntcore:ntcore-java:4.0.0
+--- edu.wpi.first.wpiutil:wpiutil-java:3.0.0
+--- org.opencv:opencv-java:3.2.0
+--- edu.wpi.first.cscore:cscore-java:1.1.0
+--- openrio.mirror.third.ctre:CTRE-phoenix-java:5.2.1.1
\--- openrio.powerup:MatchData:2018.01.07

compileOnly - Compile only dependencies for source set 'main'.
No dependencies

default - Configuration for default artifacts.
+--- edu.wpi.first.wpilibj:wpilibj-java:2018.2.2
+--- edu.wpi.first.ntcore:ntcore-java:4.0.0
+--- edu.wpi.first.wpiutil:wpiutil-java:3.0.0
+--- org.opencv:opencv-java:3.2.0
+--- edu.wpi.first.cscore:cscore-java:1.1.0
+--- openrio.mirror.third.ctre:CTRE-phoenix-java:5.2.1.1
\--- openrio.powerup:MatchData:2018.01.07

implementation - Implementation only dependencies for source set 'main'. (n)
No dependencies

nativeLib
No dependencies

nativeSimulationLib
No dependencies

nativeSimulationZip
+--- edu.wpi.first.wpilibj:wpilibj-jniShared:2018.2.2
+--- org.opencv:opencv-jni:3.2.0
+--- org.opencv:opencv-cpp:3.2.0
+--- edu.wpi.first.hal:hal:2018.2.2
+--- edu.wpi.first.wpiutil:wpiutil-cpp:3.0.0
+--- edu.wpi.first.ntcore:ntcore-cpp:4.0.0
\--- edu.wpi.first.cscore:cscore-cpp:1.1.0

nativeZip
+--- edu.wpi.first.wpilibj:wpilibj-jniShared:2018.2.2
+--- org.opencv:opencv-jni:3.2.0
+--- org.opencv:opencv-cpp:3.2.0
[... 147 lines cut ...]
+--- edu.wpi.first.wpiutil:wpiutil-java:3.0.0
+--- org.opencv:opencv-java:3.2.0
+--- edu.wpi.first.cscore:cscore-java:1.1.0
+--- openrio.mirror.third.ctre:CTRE-phoenix-java:5.2.1.1
\--- openrio.powerup:MatchData:2018.01.07

testRuntimeClasspath - Runtime classpath of source set 'test'.
+--- edu.wpi.first.wpilibj:wpilibj-java:2018.2.2
+--- edu.wpi.first.ntcore:ntcore-java:4.0.0
+--- edu.wpi.first.wpiutil:wpiutil-java:3.0.0
+--- org.opencv:opencv-java:3.2.0
+--- edu.wpi.first.cscore:cscore-java:1.1.0
+--- openrio.mirror.third.ctre:CTRE-phoenix-java:5.2.1.1
\--- openrio.powerup:MatchData:2018.01.07

testRuntimeOnly - Runtime only dependencies for source set 'test'. (n)
No dependencies

wpiTools
+--- edu.wpi.first.wpilib:SmartDashboard:2.0.4
\--- edu.wpi.first.shuffleboard:app:1.2.0

BUILD SUCCESSFUL in 3s
1 actionable task: 1 executed
```

19lmyers, 2018-02-14:
This is with WPILib set to 2018.3.1

19lmyers, 2018-02-17:
Any status on this? I'd like to make sure my team's robot is running the correct version before we bag it.

JacisNonsense, 2018-02-17:
I will take a look tonight

JacisNonsense, 2018-02-17:
I can't seem to reproduce this with GradleRIO 2018.1.22, make sure you are up to date. What platform are you running on?
You have to run at least one build connected to the internet in order to update the dependency version (`./gradlew build`).

Also, you can try stopping the daemon (`./gradlew --stop`) and rebuilding clean (`./gradlew clean build --refresh-dependencies`) if you're still finding issues.

19lmyers, 2018-02-17:
I'm running on Windows 10 1803 (17101), and I'm still having this problem after trying both those troubleshooting steps. 
```gradlew dependencies``` still prints 2018.2.2.

JacisNonsense, 2018-02-17:
I can't reproduce with the same environment. New WPI versions are now in 2018.02.17

19lmyers, 2018-03-07:
I fixed this error after realizing I have to put the wpi block BEFORE the dependencies block in my build.gradle. Now I feel silly...

commit f39442e5 (2018-01-21): Fix #73, update WPILib/SB/SD. Use GradleRIO 2018.01.21