wpilibsuite/GradleRIO #74: Fresh copy of Robot.java does not run

suclearnub, 2018-01-27:
```
Exception in thread "main" java.lang.UnsatisfiedLinkError: no wpilibJNI in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at edu.wpi.first.wpilibj.hal.JNIWrapper.<clinit>(JNIWrapper.java:58)
	at edu.wpi.first.wpilibj.RobotBase.initializeHardwareConfiguration(RobotBase.java:173)
	at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:192)

Process finished with exit code 1
```

Here's what Robo.java looks like, it has nothing in it so far:

```Java
package frc.team7242.robot;

import edu.wpi.first.wpilibj.IterativeRobot;

public class Robot extends IterativeRobot {
    @Override
    public void robotInit() { }

    @Override
    public void disabledInit() { }

    @Override
    public void autonomousInit() { }

    @Override
    public void teleopInit() { }

    @Override
    public void testInit() { }

    @Override
    public void disabledPeriodic() { }
    
    @Override
    public void autonomousPeriodic() { }

    @Override
    public void teleopPeriodic() { }

    @Override
    public void testPeriodic() { }
}
```

msoucy, 2018-01-27:
This is what I've been toying with for a few days - there currently isn't any way to run robot code locally through GradleRIO

andrewda, 2018-01-27:
Try running the `deployNativeLibs` and `deployNativeZips` Gradle tasks.

JacisNonsense, 2018-01-27:
@msoucy Define 'run locally'?

JacisNonsense, 2018-01-27:
@suclearnub can you attach your build.gradle also?

msoucy, 2018-01-28:
By locally, I meant running it as a simulation on a laptop

suclearnub, 2018-01-28:
@msoucy is the problem that I'm not deploying the code onto the robot?

suclearnub, 2018-01-28:
@JacisNonsense 
```
plugins {
    id "java"
    id "eclipse"
    id "idea"
    id "jaci.openrio.gradle.GradleRIO" version "2018.01.22"
}

def TEAM = 7242
def ROBOT_CLASS = "frc.team7242.robot.Robot"

// Define my targets (RoboRIO) and artifacts (deployable files)
// This is added by GradleRIO's backing project EmbeddedTools.
deploy {
    targets {
        target("roborio", jaci.openrio.gradle.frc.RoboRIO) {
            team = TEAM
        }
    }
    artifacts {
        artifact('frcJava', jaci.openrio.gradle.frc.FRCJavaArtifact) {
            targets << "roborio"
        }
    }
}

// Defining my dependencies. In this case, WPILib (+ friends), CTRE Phoenix (Talon SRX)
// and NavX.
dependencies {
    compile wpilib()
    compile ctre()
    compile navx()
    compile openrio.powerup.matchData()
}

// Setting up my Jar File. In this case, adding all libraries into the main jar ('fat jar')
// in order to make them all available at runtime. Also adding the manifest so WPILib
// knows where to look for our Robot Class.
jar {
    from configurations.compile.collect { it.isDirectory() ? it : zipTree(it) }
    manifest jaci.openrio.gradle.GradleRIOPlugin.javaManifest(ROBOT_CLASS)
}

task wrapper(type: Wrapper) {
    gradleVersion = '4.4'
}
```

JacisNonsense, 2018-01-28:
@suclearnub is your RoboRIO updated to 2018 Firmware and Image (v16)?

JacisNonsense, 2018-01-28:
As in you're trying to run a simulation? This isn't supported until 2019

suclearnub, 2018-01-28:
Ah, I'm not supposed to be able to run it on my machine? I'm guessing I need to plug the RoboRIO in or something to deploy it?

JacisNonsense, 2018-01-28:
Yes, you need to deploy the code to the actual robot for it to run.

closed 2018-01-28 by JacisNonsense