wpilibsuite/GradleRIO #77: UnknownHostException when discovering RoboRIO with IPv6

bherbst, 2018-02-15:
We have had intermittent issues with the `discoverRoboRio` task failing in situations where we can otherwise communicate with the RoboRIO. Often retrying a handful of times would often work

We had a laptop (running Windows 10) today that consistently had this issue, and I I think I tracked down the issue.

Looking at the debug logs when running `discoverRoboRio`, I saw that the task looking for the RoboRIO at `roborio-[TEAM]-FRC.local` was throwing an UnknownHostException on `InetAddress#getAllByName()`.

TJava will apparently prefer the IPv6 stack over the IPv4 stack when possible. (this StackOverflow post](
https://stackoverflow.com/a/1609232) led me to believe that there might be a bug with Java's IPv6 stack, but suspicion is that this might be because the RoboRIO doesn't have an IPv6 address.

Running `gradlew discoverRoboRio -Djava.net.preferIPv4Stack=true` succeeded.

Being at the end of build season, I haven't spent more time testing my theory. We also disabled IPv6 on that network adapter in Windows, which I am hoping is a more long-term fix (though we'd ideally like to re-enable that at some point).

JacisNonsense, 2018-02-15:
IPv6 has historically been a problem with mDNS resolution as far as I can remember. I'll investigate this further, but I do remember this working with IPv6 at some stage.

RIOLog uses the same method of IP lookup for the RoboRIO. If you have eclipse installed, can you check that the WPI RIOLog also fails with the same error?

bherbst, 2018-02-15:
I don't currently have Eclipse installed, but I'll plan on installing it and checking out RIOLog the next time I have a chance.

JacisNonsense, 2018-06-24:
Any update on this?

JacisNonsense, 2018-09-24:
This should be fixed now, with more discovery techniques, including DS checking

closed 2018-09-24 by JacisNonsense