wpilibsuite/GradleRIO #79: Upgrade to wpilib 3.3

tombot, 2018-02-23:
http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilibj/wpilibj-java/2018.3.3/

andrewda, 2018-02-24:
CTRE should also be updated to `5.2.2.0` as per https://firstfrc.blob.core.windows.net/frc2018/Manual/TeamUpdates/TeamUpdate14.pdf

JacisNonsense, 2018-02-24:
I'm reluctant to make any new releases after bag-day for stability reasons, as from now until April is usually OpenRIO's "lock-in period". 

I know WPILib releases are tested thoroughly automatically, but I can see no indication that the new CTRE release is stable given it was released after bag-day, is not up to date on git, and is untestable on my end since our robot is bagged.

If anyone has a practice/non-bagged robot handy, can you help test the stability of Phoenix 5.2.2.0 and WPILib 2018.3.3 before I push a change upstream?

If you want to update nevertheless, you can update the WPILib / CTRE versions in your build.gradle using the following. Make sure you run `./gradlew build` before going to your first event so you're not having to connect to the internet to download a new release.
```gradle
wpi {
    wpilibVersion = '2018.3.3'
    ctreVersion = '5.2.2.0'
}
```

andrewda, 2018-02-24:
Will test it out tomorrow. Though trying to build with `ctreVersion = '5.2.2.0'` gave me some errors about not being able to find the requirement, guessing it needs to be added to https://github.com/Open-RIO/Maven-Mirror?

JacisNonsense, 2018-02-24:
Yes, that's correct. The Maven Mirror should now be up to date and ready to pull from.

auscompgeek, 2018-02-24:
Note that the Phoenix 5.2.2.0 changes specifically affect Java only.

JacisNonsense, 2018-02-24:
That is true, although in GradleRIO the C++ and Java dependency versions are tied.

bdaroz, 2018-02-26:
I know the CTRE libs have been somewhat... controversial... this year, but unless I'm mistaken, we need to manually enter the version of GradeRIO to use (if we followed the examples). What's the harm in continuing to track and update the versions in GradeRIO instead of teams handing one-offs?

JacisNonsense, 2018-02-26:
I would be more than comfortable pushing an update during the season, but since robots are in bags now, I don't want to push any updates unless I know they work (I call this the OpenRIO 'lock-in period'). The main reason is so that if a team changes their GradleRIO version, then it stops working at competition because of some mismatch, they don't need to downgrade that master version (which may or may not require an internet connection).

I guess another way to phrase this is that I want it to be **explicitly** known _what_ changes are happening between bag-day and comp-day by the team, not abstracted behind some big version number in case there is a breaking change in one of those dependencies.

JacisNonsense, 2018-02-26:
As a footnote, the reason I am asking for CTRE to be tested specifically is because their Git is not up to date, nor are there any public tests, so I can't verify there aren't any major breaking changes. With WPILib, I can see and verify the changes and test results.

bdaroz, 2018-02-26:
I agree w/ the CTRE stuff - when we get our practice bot rolling again this week we'll be able to test it then. In the meanwhile I opened up an issue on the CTRE public repo about it.

andrewda, 2018-02-26:
We tested `v5.2.2.0` a little on our practice bot yesterday and it seemed to work fine without any hiccups for the 30+ minutes we were running it.

tombot, 2018-02-26:
We used 5.2.2.0 and wpilib 2018.3.3 all weekend on our practice robot. The part of the API we utilize seemed ok.

bdaroz, 2018-02-27:
Our build team decided today was a good day to tear up the practice bot to get it under the weight limit (and match comp bot) so I didn't get to test this today. Next possible is Wed.

In other news: CrossTheRoadElec/Phoenix-frc-lib#34 has been closed and the repo updated, such that it is.

JacisNonsense, 2018-03-01:
I'm putting this off for a few days since there's a pending NTCore release coming sometime in the next week or so. CTRE also now has 5.3.1.0

JacisNonsense, 2018-03-03:
Update: the version of Phoenix that I push will be 5.2.2.0. Seems 5.3.1.0 has breaking changes on the simplest of codebases. 5.2.2.0, however, seems stable enough with no breaking changes.

Thanks for your patience everyone, just want to make 110% sure we're not breaking peoples code at this time, we don't want to be responsible for a team missing a match.

JacisNonsense, 2018-03-03:
Release seems on track for Monday AWST (Sunday EST)

CardcaptorRLH85, 2018-03-04:
While I'd like to see where I can find the actual changelog for Phoenix (more so than the release notes that come with the installer), I just wanted to say that our team used 5.3.1.0 without having to make any changes around it at our week one competition (Kettering District #1).

andrewda, 2018-03-04:
Yes, we're using 5.3.1.0 without having to make any other changes. All for not having it as a default in GradleRIO if it has breaking changes, though.

JacisNonsense, 2018-03-04:
It would seem the breaking changes only affect the C++ version of Phoenix.

JacisNonsense, 2018-03-04:
See here: https://github.com/CrossTheRoadElec/Phoenix-frc-lib/commit/e8221da18ba905ddddc20b5eee99d0ed31d070a0
It's minor but not something I'm willing to risk changing for teams with prebuilt libs. I'll push 5.2.2.0 but you can manually set the below flag if you're comfortable updating. Update should be coming tomorrow. Thanks again
```gradle
wpi {
    ctreVersion = '5.3.1.0'
}
```

JacisNonsense, 2018-03-06:
Thanks for your patience everyone, the new update is now out. Update to GradleRIO `2018.03.06` to use the new versions of WPILib, NTCore, CSCore, and WPIUtil published last night, and CTRE 5.2.2.0.

commit 3126b14d (2018-03-06): Update dependencies. Fixes #79
closed 2018-03-06 by JacisNonsense