wpilibsuite/GradleRIO #81: NullPointerException in discoverRoborio

bdaroz, 2018-03-09:
Getting an NPE when attempting to deploy on one of our computers....

```
Caused by: java.lang.NullPointerException
 	at jaci.gradle.deploy.tasks.TargetDiscoveryTask$_discoverTarget_closure3.doCall(TargetDiscoveryTask.groovy:105)
```

Full Debug

```
(Snip)
13:43:02.313 [DEBUG] [org.gradle.internal.progress.DefaultBuildOperationExecutor] Completing Build operation 'Resolve wpilibc-linuxathena.zip (edu.wpi.first.wpilibc:wpilibc:2018.4.1)'
13:43:02.313 [DEBUG] [org.gradle.launcher.daemon.server.SynchronizedDispatchConnection] thread 435: dispatching class org.gradle.launcher.daemon.protocol.BuildEvent
13:43:02.313 [DEBUG] [org.gradle.internal.progress.DefaultBuildOperationExecutor] Build operation 'Resolve wpilibc-linuxathena.zip (edu.wpi.first.wpilibc:wpilibc:2018.4.1)' completed
13:43:02.313 [DEBUG] [org.gradle.internal.work.DefaultWorkerLeaseService] Worker lease root.1.103.104 completed (1 worker(s) in use)
13:43:02.313 [DEBUG] [org.gradle.internal.resources.AbstractTrackedResourceLock] Daemon worker Thread 8: released lock on root.1.103.104
13:43:02.313 [DEBUG] [org.gradle.internal.progress.DefaultBuildOperationExecutor] Completing Build operation 'Resolve files of :native_wpilibc_athena'
13:43:02.313 [DEBUG] [org.gradle.launcher.daemon.server.SynchronizedDispatchConnection] thread 435: dispatching class org.gradle.launcher.daemon.protocol.BuildEvent
13:43:02.313 [DEBUG] [org.gradle.internal.progress.DefaultBuildOperationExecutor] Build operation 'Resolve files of :native_wpilibc_athena' completed
13:43:02.313 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_athena.sharedMatchers' from state GraphClosed to Initialized
13:43:02.313 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_athena.staticMatchers' from state GraphClosed to Initialized
13:43:02.313 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_athena.libraryMatchers' from state GraphClosed to Initialized
13:43:02.313 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_athena.headerDirs' from state GraphClosed to Initialized
13:43:02.313 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_athena.libraryNames' from state GraphClosed to Initialized
13:43:02.313 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_headers' from state GraphClosed to Initialized
13:43:02.314 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_headers' from state GraphClosed to Discovered
13:43:02.314 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_headers' from state GraphClosed to Discovered
13:43:02.314 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_headers.name' from state GraphClosed to Initialized
13:43:02.314 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_headers.mainLibraryName' from state GraphClosed to Initialized
13:43:02.314 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_headers.flavor' from state GraphClosed to Initialized
13:43:02.314 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_headers.buildType' from state GraphClosed to Initialized
13:43:02.314 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_headers.targetPlatforms' from state GraphClosed to Initialized
13:43:02.314 [DEBUG] [org.gradle.model.internal.registry.DefaultModelRegistry] Project : - Transitioning model element 'libraries.wpilibc_headers.maven' from state GraphClosed to Initialized
13:43:02.314 [DEBUG] [org.gradle.launcher.daemon.server.SynchronizedDispatchConnection] thread 435: dispatching class org.gradle.launcher.daemon.protocol.BuildEvent
13:43:02.314 [DEBUG] [org.gradle.internal.progress.DefaultBuildOperationExecutor] Build operation 'Resolve dependencies of :native_wpilibc_headers' started
13:43:02.314 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine.DefaultArtifactDependencyResolver] Resolving configuration ':native_wpilibc_headers'
13:43:02.314 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.ivyresolve.memcache.InMemoryCachedRepositoryFactory] Reusing in-memory cache for repo 'WPI' [a7de36dd177eef9716c1873cc534d523].
13:43:02.314 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.ivyresolve.memcache.InMemoryCachedRepositoryFactory] Reusing in-memory cache for repo 'Jaci' [e0958b754a0400bde8048a4cf97fcf5b].
13:43:02.314 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.ivyresolve.memcache.InMemoryCachedRepositoryFactory] Reusing in-memory cache for repo 'OpenRIO' [90c3b24590b6a4068c5250b875213ff9].
13:43:02.315 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.ivyresolve.memcache.InMemoryCachedRepositoryFactory] Reusing in-memory cache for repo 'MavenRepo' [3bfc68f2a4d6150b392f01d68a2c9470].
13:43:02.315 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine.graph.builder.DependencyGraphBuilder] Visiting configuration org.techvalleyhigh.frc5881.powerup:robot:1.0-SNAPSHOT(native_wpilibc_headers).
13:43:02.315 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.ivyresolve.RepositoryChainComponentMetaDataResolver] Attempting to resolve component for edu.wpi.first.wpilibc:wpilibc:2018.4.1 using repositories [WPI, Jaci, OpenRIO, MavenRepo]
13:43:02.315 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.ivyresolve.RepositoryChainComponentMetaDataResolver] Using edu.wpi.first.wpilibc:wpilibc:2018.4.1 from Maven repository 'WPI'
13:43:02.315 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine.graph.builder.DependencyGraphBuilder] Visiting configuration edu.wpi.first.wpilibc:wpilibc:2018.4.1(default).
13:43:02.315 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine.graph.builder.DependencyGraphBuilder] edu.wpi.first.wpilibc:wpilibc:2018.4.1(default) has no transitive incoming edges. ignoring outgoing edges.
13:43:02.315 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine.oldresult.TransientConfigurationResultsBuilder] Flushing resolved configuration data in Binary store in /private/var/folders/9t/tmrp6g117zs5yf52jnq_dr400000gn/T/gradle5964781979915301344.bin. Wrote root 2.
13:43:02.315 [DEBUG] [org.gradle.internal.progress.DefaultBuildOperationExecutor] Completing Build operation 'Resolve dependencies of :native_wpilibc_headers'
13:43:02.315 [DEBUG] [org.gradle.launcher.daemon.server.SynchronizedDispatchConnection] thread 435: dispatching class org.gradle.launcher.daemon.protocol.BuildEvent
13:43:02.315 [DEBUG] [org.gradle.internal.progress.DefaultBuildOperationExecutor] Build operation 'Resolve dependencies of :native_wpilibc_headers' completed
13:43:02.315 [DEBUG] [org.gradle.internal.component.model.ComponentAttributeMatcher] Selected match edu.wpi.first.wpilibc:wpilibc:2018.4.1 from candidates [edu.wpi.first.wpilibc:wpilibc:2018.4.1] for {}
13:43:02.315 [DEBUG] [org.gradle.launcher.daemon.server.SynchronizedDispatchConnection] thread 435: dispatching class org.gradle.launcher.daemon.protocol.BuildEvent
13:43:02.315 [DEBUG] [org.gradle.internal.progress.DefaultBuildOperationExecutor] Build operation 'Resolve files of :native_wpilibc_headers' started
13:43:02.315 [DEBUG] [org.gradle.internal.component.model.ComponentAttributeMatcher] Selected match edu.wpi.first.wpilibc:wpilibc:2018.4.1 from candidates [edu.wpi.first.wpilibc:wpilibc:2018.4.1] for {}
13:43:02.316 [DEBUG] [org.gradle.api.internal.artifacts.ivyservice.resolveengine.oldresult.TransientConfigurationResultsBuilder] Loaded resolved configuration results (0.0 secs) from Binary store in /private/var/folders/9t/tmrp6g117zs5yf52jnq_dr400000gn/T/gradle5964781979915301344.bin
[... cut ...]
13:43:03.786 [DEBUG] [org.gradle.cache.internal.btree.BTreePersistentIndexedCache] Closing cache module-metadata.bin (/Users/brianr/.gradle/caches/modules-2/metadata-2.36/module-metadata.bin)
13:43:03.787 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on artifact cache (/Users/brianr/.gradle/caches/modules-2).
13:43:03.787 [DEBUG] [org.gradle.launcher.daemon.server.exec.ExecuteBuild] The daemon has finished executing the build.
Execution failed for task ':discoverRoborio'.
1:43:03 PM: Task execution finished 'discoverRoborio --stacktrace --debug'.
```

JacisNonsense, 2018-03-20:
What does your build.gradle look like?

tylerscheuble, 2018-03-22:
We've had this issue happen to us twice now. Restarting the deploying computer made the issue go away both times.

JacisNonsense, 2018-03-23:
Sure the issue also resolve itself by running `./gradlew deploy --no-daemon`?

JacisNonsense, 2018-06-24:
Any update on this?

bdaroz, 2018-06-24:
When it happened I gathered the debug output and went to a backup computer to build on. (We were at comp - I opened this after the event.) We haven't had it happen since, or before we got to comp. Sorry for the dead-air I had completely forgotten I had opened this issue.

For the record our `build.gradle` is almost exactly like the sample, but I can get you access to it if you need it.

At this point our robot is on it's summer tour and I won't get hands-on with it again until September. :(