wpilibsuite/GradleRIO #82: Execution failed for task ':deployFrcNative'

StripedMonkey, 2018-03-18:
During Comps I discovered this issue and ended up having to use eclipse to deploy.
[ErrorGradleRIO.txt](https://github.com/Open-RIO/GradleRIO/files/1823010/ErrorGradleRIO.txt)

Any clues as to who/what/why? This is an issue only with deploying.

JacisNonsense, 2018-03-20:
Please run with `--debug` and send the full log. 

Have you run on MINGW before this? We don't officially support it

JacisNonsense, 2018-03-21:
Can you list the files present in `C:\Users\HP Owner\Projects\20186214\build\exe\frcUserProgram\` with `ls -al` ?

StripedMonkey, 2018-03-21:
It appears to be empty.
```bash
 ls -al build/exe/frcUserProgram/
total 0
drwxr-xr-x 1 HP Owner 197121 0 Mar 21 16:17 ./
drwxr-xr-x 1 HP Owner 197121 0 Mar 21 16:17 ../
```

StripedMonkey, 2018-03-21:
I just tried it with powershell and both the gradlew & gradlew.bat and nether appear to work.

JacisNonsense, 2018-03-22:
You've run `./gradlew installToolchain`, right?

StripedMonkey, 2018-03-22:
Correct

StripedMonkey, 2018-03-22:
There *is* a connection to the RoboRIO by the way, when deploying code using GradleRIO it will kill the code currently on the robot and will not come back until you deploy using the Eclipse Plugin

JacisNonsense, 2018-03-23:
Given a build file isn't being generated, this sounds like an issue with MINGW. Run ./gradlew build --debug --console=plain and send me the output

StripedMonkey, 2018-03-23:
Tbh I'm not sure why that would be. It compiles fine and deploys without issue using the eclipse plugin. Additionally (you know far more than me btw, this is all me trying to understand it) doesn't the "installtoolchain" and the wpilib thing cover everything nessicary to build and compile?

I'll test it out ASAP but unfortunately comps are this weekend for me so I'll be a bit tied up.

JacisNonsense, 2018-03-23:
The toolchain isn't shared between MINGW and your windows system, it likely won't work on mingw which is why we don't officially supported. If gradle can't find the toolchain, we've told it to skip the build.

The wpilib toolchain and `installToolchain` install the toolchain to different directories. If you want to build and deploy, don't use mingw since it probably won't work, use cmd/powershell instead

StripedMonkey, 2018-03-23:
I've tried Powershell though and it gave the same error.

JacisNonsense, 2018-03-24:
In that case, try `./gradlew components` and show me the output. For some reason, it isn't producing a binary

StripedMonkey, 2018-03-24:
I managed to figure out the issue, apparently I don't have my second directories properly configured. 

I don't have my program structured in the src/cpp src/headers format.

closed 2018-03-24 by JacisNonsense