wpilibsuite/GradleRIO #89: ./gradlew deploy doesn't find roboRIO

landonh12, 2018-05-08:
I've been trying to deploy code to my roboRIO using GradleRIO. I've tried 2 different computers, macOS and Windows, WiFi and USB on both. I get the same output ever time. Here's the output when I run `./gradlew deploy --stacktrace --info --offline`

`Landons-MacBook-Pro:2018java landonhaugh$ ./gradlew deploy --stacktrace --info --offline
Initialized native services in: /Users/landonhaugh/.gradle/native
The client will now receive all logging from the daemon (pid: 574). The daemon log file: /Users/landonhaugh/.gradle/daemon/4.4/daemon-574.out.log
Starting 6th build in daemon [uptime: 4 mins 47.091 secs, performance: 100%, no major garbage collections]
Using 4 worker leases.
Creating new cache for fileHashes, path /Users/landonhaugh/2018java/.gradle/4.4/fileHashes/fileHashes.bin, access org.gradle.cache.internal.DefaultCacheAccess@31dd7e7
Starting Build
Settings evaluated using settings file '/Users/landonhaugh/2018java/settings.gradle'.
Projects loaded. Root project using build file '/Users/landonhaugh/2018java/build.gradle'.
Included projects: [root project '2018java']

> Configure project : 
Evaluating root project '2018java' using build file '/Users/landonhaugh/2018java/build.gradle'.
Creating new cache for metadata-1.1/results, path /Users/landonhaugh/.gradle/caches/transforms-1/metadata-1.1/results.bin, access org.gradle.cache.internal.DefaultCacheAccess@47e37ef2
Creating new cache for metadata-2.36/module-metadata, path /Users/landonhaugh/.gradle/caches/modules-2/metadata-2.36/module-metadata.bin, access org.gradle.cache.internal.DefaultCacheAccess@545c3a29
Creating new cache for metadata-2.36/module-artifacts, path /Users/landonhaugh/.gradle/caches/modules-2/metadata-2.36/module-artifacts.bin, access org.gradle.cache.internal.DefaultCacheAccess@545c3a29
Creating new cache for metadata-2.36/module-artifact, path /Users/landonhaugh/.gradle/caches/modules-2/metadata-2.36/module-artifact.bin, access org.gradle.cache.internal.DefaultCacheAccess@545c3a29

All projects evaluated.
Selected primary task 'deploy' from project :
Creating new cache for annotation-processors, path /Users/landonhaugh/2018java/.gradle/4.4/fileContent/annotation-processors.bin, access org.gradle.cache.internal.DefaultCacheAccess@7d2d69da
Tasks to be executed: [task ':discoverRoborio', task ':compileJava', task ':processResources', task ':classes', task ':jar', task ':deployFrcJava', task ':deployJre', task ':deployNativeLibs', task ':deployNativeZips', task ':deployRoborioCommands', task ':deploy']
Creating new cache for resourceHashesCache, path /Users/landonhaugh/2018java/.gradle/4.4/fileHashes/resourceHashesCache.bin, access org.gradle.cache.internal.DefaultCacheAccess@31dd7e7
Creating new cache for taskHistory, path /Users/landonhaugh/2018java/.gradle/4.4/taskHistory/taskHistory.bin, access org.gradle.cache.internal.DefaultCacheAccess@3feff6d4
Creating new cache for outputFiles, path /Users/landonhaugh/2018java/.gradle/buildOutputCleanup/outputFiles.bin, access org.gradle.cache.internal.DefaultCacheAccess@5d248c87
:discoverRoborio (Thread[Task worker for ':',5,main]) started.
Resolved 10.3.64.2 -> 10.3.64.2
Resolved 172.22.11.2 -> 172.22.11.2
Resolved roborio-364-FRC.local -> 10.3.64.52

> Task :discoverRoborio 
Putting task artifact state for task ':discoverRoborio' into context took 0.0 secs.
Up-to-date check for task ':discoverRoborio' took 0.0 secs. It is not up-to-date because:
  Task has not declared any outputs.
Discovering Target roborio

:discoverRoborio (Thread[Task worker for ':',5,main]) completed. Took 3.513 secs.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':discoverRoborio'.
> Target roborio could not be located! Failing as roborio.failOnMissing is true.

* Try:
Run with --debug option to get more log output. Run with --scan to get full insights.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':discoverRoborio'.
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:100)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:70)
        at org.gradle.api.internal.tasks.execution.OutputDirectoryCreatingTaskExecuter.execute(OutputDirectoryCreatingTaskExecuter.java:51)
        at org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter.execute(SkipUpToDateTaskExecuter.java:62)
        at org.gradle.api.internal.tasks.execution.ResolveTaskOutputCachingStateExecuter.execute(ResolveTaskOutputCachingStateExecuter.java:54)
        at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.execute(ValidatingTaskExecuter.java:60)
        at org.gradle.api.internal.tasks.execution.SkipEmptySourceFilesTaskExecuter.execute(SkipEmptySourceFilesTaskExecuter.java:97)
        at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:87)
        at org.gradle.api.internal.tasks.execution.ResolveTaskArtifactStateTaskExecuter.execute(ResolveTaskArtifactStateTaskExecuter.java:52)
        at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:52)
        at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:54)
        at org.gradle.api.internal.tasks.execution.ExecuteAtMostOnceTaskExecuter.execute(ExecuteAtMostOnceTaskExecuter.java:43)
        at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:34)
        at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker$1.run(DefaultTaskGraphExecuter.java:248)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:336)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:328)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:199)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:110)
        at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter$EventFiringTaskWorker.execute(DefaultTaskGraphExecuter.java:241)
[... 13 lines cut ...]
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:73)
        at org.gradle.api.internal.project.taskfactory.StandardTaskAction.doExecute(StandardTaskAction.java:46)
        at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:39)
        at org.gradle.api.internal.project.taskfactory.StandardTaskAction.execute(StandardTaskAction.java:26)
        at org.gradle.api.internal.AbstractTask$TaskActionWrapper.execute(AbstractTask.java:780)
        at org.gradle.api.internal.AbstractTask$TaskActionWrapper.execute(AbstractTask.java:747)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter$1.run(ExecuteActionsTaskExecuter.java:121)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:336)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:328)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:199)
        at org.gradle.internal.progress.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:110)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeAction(ExecuteActionsTaskExecuter.java:110)
        at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:92)
        ... 29 more

* Get more help at https://help.gradle.org

BUILD FAILED in 4s
1 actionable task: 1 executed
Telemetry skipped! (no-telemetry = false, ReportTelemetry = true, offline = true, CI = false) (expected false, true, false, false)`

What's going on here?

edelmanjm, 2018-05-09:
Perhaps I'm misreading the stacktrace, but that sounds like the RoboRIO isn't updated correctly. What image is it running?

JacisNonsense, 2018-05-09:
From your chief post it looks like you've solved this already. For those reading at a later date: labview was pinning RoboRIO CPU and as such the SSH server couldn't accept a new connection

closed 2018-05-09 by JacisNonsense