wpilibsuite/GradleRIO #112: Use Kauai Labs' official Maven repository

RobinsonZ, 2018-05-25:
Per [this page](https://www.pdocs.kauailabs.com/navx-mxp/software/roborio-libraries/java/), there's an official Kauai Labs Maven repository at http://www.kauailabs.com/maven2 with the NavX artifact (at least for Java) at `com.kauailabs.navx.frc:navx_frc`. Probably would be good to use that instead of the OpenRIO mirror.

edelmanjm, 2018-05-25:
y tho

JacisNonsense, 2018-05-26:
We were working on switching to KL's official repo earlier this year, however since C++ releases weren't published, we didn't want to create a possible disparity between Java and C++ release schedules.

We're working on changing this for 2019.

JacisNonsense, 2018-09-23:
Closed since this is a vendor library issue now (we no longer package NavX and CTRE releases, they are added by including a `.json` file in your project directory, which is managed by VSCode / can be done very easily manually)

closed 2018-09-23 by JacisNonsense