wpilibsuite/GradleRIO #122: C++ Mac Builds Broken

AustinShalit, 2018-07-19:
I can't seem to build C++ projects on Mac.

See error below:
```
See file:///Users/austinshalit/Desktop/TestCpp/build/tmp/linkFrcUserProgramExecutable/output.txt for all output for linkFrcUserProgramExecutable.
linking frcUserProgram failed.
/usr/local/lib/gcc/arm-frc-linux-gnueabi/5.5.0/../../../../arm-frc-linux-gnueabi/bin/ld: cannot find crt1.o: No such file or directory
/usr/local/lib/gcc/arm-frc-linux-gnueabi/5.5.0/../../../../arm-frc-linux-gnueabi/bin/ld: cannot find crti.o: No such file or directory
/usr/local/lib/gcc/arm-frc-linux-gnueabi/5.5.0/../../../../arm-frc-linux-gnueabi/bin/ld: cannot find -lm
collect2: error: ld returned 1 exit status

Finished linkFrcUserProgramExecutable, see full log file:///Users/austinshalit/Desktop/TestCpp/build/tmp/linkFrcUserProgramExecutable/output.txt.
```

JacisNonsense, 2018-07-20:
I can't seem to reproduce this, works on my machine^TM. 

Can you run `./gradlew explainToolchains`?

JacisNonsense, 2018-07-20:
Also link me your project, if you can

AustinShalit, 2018-07-20:
Project is the TimedRobot template.

```
> Task :explainToolchains
FRC Home:
  - Valid?: false
  - Found?: false
  - Version Range:
      - Low: 5.5
      - High: 5.5
      - Is Valid?: false
  - Root: null
  - Bin: null
  - Lib: null
  - Include: null
  - Gcc: null
  - Gdb: null
GradleRIO Home:
  - Valid?: false
  - Found?: false
  - Version Range:
      - Low: 5.5
      - High: 5.5
      - Is Valid?: false
  - Root: null
  - Bin: null
  - Lib: null
  - Include: null
  - Gcc: null
  - Gdb: null
System Path 0:
  - Valid?: true
  - Found?: true
  - Version Range:
      - Low: 5.5
      - High: 5.5
      - Is Valid?: true
  - Root: /usr/local
  - Bin: /usr/local/bin
  - Lib: /usr/local/lib
  - Include: /usr/local/include
  - Gcc: /usr/local/bin/arm-frc-linux-gnueabi-g++
  - Gdb: /usr/local/bin/arm-frc-linux-gnueabi-gdb
  - Metadata:
      - Version: 5.5.0
      - Vendor: gcc version 5.5.0 (GCC for FRC)
      - Default Arch: architecture 'x86-64'
      - System Libraries:
          - Include:
              - /usr/local/lib/gcc/arm-frc-linux-gnueabi/5.5.0/include
              - /usr/local/lib/gcc/arm-frc-linux-gnueabi/5.5.0/include-fixed
              - /usr/local/lib/gcc/arm-frc-linux-gnueabi/5.5.0/../../../../arm-frc-linux-gnueabi/include
              - /usr/local/arm-frc-linux-gnueabi/usr/include
          - Lib Dirs
          - Macros

Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.
See https://docs.gradle.org/4.8/userguide/command_line_interface.html#sec:command_line_warnings
```

ThadHouse, 2018-07-20:
Can you upload the following file austin?

`/Users/austinshalit/Desktop/TestCpp/build/tmp/linkFrcUserProgramExecutable/options.txt`

AustinShalit, 2018-07-20:
>/Users/austinshalit/Desktop/TestCpp/build/tmp/linkFrcUserProgramExecutable/options.txt
```
-o
/Users/austinshalit/Desktop/TestCpp/build/exe/frcUserProgram/frcUserProgram
/Users/austinshalit/Desktop/TestCpp/build/objs/frcUserProgram/frcUserProgramCpp/8mxatcv8fo1re5py2g1gv6aur/ExampleSubsystem.o
/Users/austinshalit/Desktop/TestCpp/build/objs/frcUserProgram/frcUserProgramCpp/bnbreoh6cpujbjqqqu41gfy99/ExampleCommand.o
/Users/austinshalit/Desktop/TestCpp/build/objs/frcUserProgram/frcUserProgramCpp/9760xtgdau9jsoxxi90ozrpww/Robot.o
/Users/austinshalit/Desktop/TestCpp/build/objs/frcUserProgram/frcUserProgramCpp/565zo23b3cn3z31bfeb83vj15/OI.o
/Users/austinshalit/Desktop/TestCpp/build/objs/frcUserProgram/frcUserProgramCpp/8fqvn0jrx7w4imyb7ld326f0s/MyAutoCommand.o
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/wpilibc-2018.4.1-linuxathena.zip_271aadd5fe68d534a7ec7d8bcf00a5f6/linux/athena/shared/libwpilibc.so
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/hal-2018.4.1-linuxathena.zip_ec5b4e6116b39724b13940cebe9e01a1/linux/athena/shared/libwpiHal.so
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/wpiutil-cpp-3.2.0-linuxathena.zip_b3c0854cbf80984db4dae519ac88ea3b/linux/athena/shared/libwpiutil.so
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/ntcore-cpp-4.1.0-linuxathena.zip_25c0a1cfddd66d5210163e2a58b2dd47/linux/athena/shared/libntcore.so
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/cscore-cpp-1.3.0-linuxathena.zip_5d3f92c07ea1bd9fe3d09c0df5c59501/linux/athena/shared/libcscore.so
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_calib3d.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_core.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_features2d.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_flann.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_highgui.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_imgcodecs.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_imgproc.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_java320.so
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_ml.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_objdetect.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_photo.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_shape.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_stitching.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_superres.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_video.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_videoio.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/opencv-cpp-3.2.0-linuxathena.zip_882ce6d6786024fd1378ddee15c75ec3/linux/athena/shared/libopencv_videostab.so.3.2
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/ni-libraries-2018.4.1-linuxathena.zip_2c64ecd4b3048d6e908a9a8357e6be0a/linux/athena/shared/libFRC_NetworkCommunication.so.18.0.0
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/ni-libraries-2018.4.1-linuxathena.zip_2c64ecd4b3048d6e908a9a8357e6be0a/linux/athena/shared/libNiFpga.so.17.0.0
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/ni-libraries-2018.4.1-linuxathena.zip_2c64ecd4b3048d6e908a9a8357e6be0a/linux/athena/shared/libNiFpgaLv.so.17.0.0
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/ni-libraries-2018.4.1-linuxathena.zip_2c64ecd4b3048d6e908a9a8357e6be0a/linux/athena/shared/libNiRioSrv.so.17.0.0
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/ni-libraries-2018.4.1-linuxathena.zip_2c64ecd4b3048d6e908a9a8357e6be0a/linux/athena/shared/libRoboRIO_FRC_ChipObject.so.18.0.0
[... cut ...]
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/ni-libraries-2018.4.1-linuxathena.zip_2c64ecd4b3048d6e908a9a8357e6be0a/linux/athena/shared/libvisa.so
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/CTRE-phoenix-cpp-5.5.1.0.zip_585f0cfd494105495284283f2448b573/libCTRE_Phoenix.a
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/CTRE-phoenix-cpp-5.5.1.0.zip_585f0cfd494105495284283f2448b573/libCTRE_PhoenixCCI.a
/Users/austinshalit/Desktop/TestCpp/build/tmp/expandedArchives/navx-cpp-3.0.348.zip_225c7dd2da6cb7be473ed62ee25c816c/libnavx_frc_cpp.a
--sysroot
/usr/local
-pthread
-rdynamic
```