wpilibsuite/GradleRIO #134: IntelliJ Idea doesn't recognize grade?

GaganBhat, 2018-07-30:


edelmanjm, 2018-07-30:
Did you initially import the project as a Gradle project or add it after the fact?

GaganBhat, 2018-07-30:
@edelmanjm sorry for the lack of description, didn't think that the enter button would post the issue instantly. I imported the project from the welcome screen, and it recognized it as a grade project, I checked auto import, and finish.

edelmanjm, 2018-07-30:
Can you post your project structure?

GaganBhat, 2018-07-30:
[image]

Here's the structure.

GaganBhat, 2018-07-30:
`
FAILURE: Build failed with an exception.

* What went wrong:
Could not create service of type ScriptPluginFactory using BuildScopeServices.createScriptPluginFactory().
> Could not create service of type CrossBuildFileHashCache using BuildSessionScopeServices.createCrossBuildFileHashCache().

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 0s
Failed to create parent directory '/Users/chandana.bhat@in.ibm.com/Documents/Programming/Java/FRC/Basebot-Code/.gradle/4.4' when creating directory '/Users/chandana.bhat@in.ibm.com/Documents/Programming/Java/FRC/Basebot-Code/.gradle/4.4/fileHashes'
`

I also get the above error when I try to refresh gradle by clicking on the sync button. 
[image]

JacisNonsense, 2018-07-30:
Your gradle version is well out of date. Update to 4.9 and try again

GaganBhat, 2018-07-30:
Alright, it was previously not using the local gradle install, so I did that now. Here's the error I am getting now. 

```

FAILURE: Build failed with an exception.

* What went wrong:
Could not create service of type ScriptPluginFactory using BuildScopeServices.createScriptPluginFactory().
> Could not create service of type FileHasher using BuildSessionScopeServices.createFileSnapshotter().

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
/Users/chandana.bhat@in.ibm.com/Documents/Programming/Java/FRC/Basebot-Code/.gradle/4.9/fileHashes/fileHashes.lock (Permission denied)
```

GaganBhat, 2018-07-30:
[image]

Another weird thing is that it builds the project if I run it from the terminal. But doesn't work with IntelliJ.

GaganBhat, 2018-07-30:
I just imported it into my windows machine, everything works perfectly. 
[image]

JacisNonsense, 2018-07-30:
You should definitely not run gradle as root. chown ~/.gradle, projectdir/gradle and projectdir/.gradle for your current user

JacisNonsense, 2018-07-30:
Run `./gradlew --version` and send me the output. You may have to update your JDK

GaganBhat, 2018-07-30:
[image]

JacisNonsense, 2018-07-30:
`--version`, not `version`

GaganBhat, 2018-07-30:
[image]

GaganBhat, 2018-07-30:
Alright, I updated project pref. to 1.8  and looks like it build successfully. Thanks for the help!

closed 2018-07-30 by GaganBhat