wpilibsuite/GradleRIO #136: Allow for the use of Gradle Kotlin DSL

armeenm, 2018-08-09:
Have a way for users wanting to use the Gradle Kotlin DSL to do so. Currently this is not practical as the dependency lambdas (e.g. `compile wpilib()`) cannot be seen due to runtime decoration.

JLLeitschuh, 2018-10-17:
I'm working on this across a few fixes in the Gradle and Gradle Kotlin DSL projects:

https://github.com/gradle/gradle/pull/7414
https://github.com/gradle/kotlin-dsl/pull/1190

I'm hoping to get these features into Gradle 5.0 which should make the Gradle Kotlin DSL much nicer.

If you really want to access the `wpilib` function currently in the Kollin DSL you can do something like this. Note, this API will probably change before kickoff, just be warned.
```kotlin
val wpilib : Closure<List<String>> by (project.dependencies as ExtensionAware).extensions 

dependencies {
    wpilib().forEach { compile(it) }
}
```
I haven't actually tried this, this is just me spitballing given my experience with the Gradle Kotlin DSL.

It would be really awesome if teams could use the Kotlin DSL. I'll see what I can do to do make it happen. No promises though.

JLLeitschuh, 2018-10-19:
One thing that needs to happen is that this logic here needs to change.

https://github.com/wpilibsuite/GradleRIO/blob/03ce5bcddfb7d9fe98fcf9b17d2da44d1e0f353f/src/main/groovy/edu/wpi/first/gradlerio/wpi/dependencies/WPIJavaDeps.groovy#L43-L68

These need to be changed.
`project.dependencies.ext.wpilib` needs to change from being a `Closure<List<String>>` to a `Callable<List<String>>`.

@ThadHouse Was tinkering with this me a few nights ago.

JacisNonsense, 2018-10-20:
I was going to move `wpilib` etc to an extension that strongly types them as functions

JacisNonsense, 2018-12-30:
This should be mostly resolved now. Are there any features of GR that still don't work in the kotlin DSL?

msoucy, 2018-12-31:
Some notes from attempting to use version `2019.1.1-beta-4-pre4`:

- Use the dependencies block (from a quick test, `compile(wpilib())` doesn't seem to find the wpilib name)
- Configure the EmbeddedTools `deploy` block, as `target("roborio", edu.wpi.first.gradlerio.frc.RoboRIO) {` now produces a lot of "expecting an element" syntax errors
- Note that `targets << "roborio"` needs to become `targets.append("roborio")` in that same block
- `getTeamNumber()` and `getDebugOrDefault()` can't be found (might be related to the first few issues)

prateekma, 2018-12-31:
I got my build.gradle.kts to fully compile with Beta 4b.

https://gist.github.com/prateekma/0f9338dd9bff35bcc3fb3559c1c266be

Starlight220, 2020-10-01:
I played with this a bit, @prateekma's `build.gradle.kts` doesn't compile with updated versions. Many problems occur from blocks with a receiver that Kotlin/IntelliJ reads as a `NamedDomainObjectCollection`, therefore needs an explicit cast to access inner functions/properties

Starlight220, 2020-10-29:
Seems that most of the Kotlin DSL problems need to be fixed from the [EmbeddedTools](https://github.com/JaciBrunning/EmbeddedTools) side. Not much can be done from this repository

prateekma, 2020-10-29:
The Kotlin Gradle DSL still works with the latest GradleRIO: https://gist.github.com/prateekma/e466d1421a128d8270f2044bac880c6d

closed 2022-01-05 by armeenm