wpilibsuite/GradleRIO #139: When I add a dependency to a local library to the build.gradle file, I get an exception when trying to build.

sjcbulldog, 2018-08-25:
Our robot code consists of a set of libraries that get reused each year, and a top level application that varies from year to year.  We build the code that is reused as libraries and link the top level application against this code.  This worked for us using the previous versions of GradleRIO, but does not work with the current version that is part of the alpha release.  

This is on a windows platform.  I have not tried on Linux or Mac OS.

This is the output of gradle with --stracktrace
[gradle.out.txt](https://github.com/wpilibsuite/GradleRIO/files/2320780/gradle.out.txt)

This is the output of gradle with --stacktrace --debug
[gradle.debug.txt](https://github.com/wpilibsuite/GradleRIO/files/2320779/gradle.debug.txt)

This file contains a sample project that demonstrates the problem.  This is the project that was run to produce the above output files.
[xyzzy.zip](https://github.com/wpilibsuite/GradleRIO/files/2320762/xyzzy.zip)

Let me know if I can be of any more help.

ThadHouse, 2018-08-25:
Can you post your project from last year as well? I seem to be able to reproduce this bug even when switching to last years GradleRIO, so I want to make sure we're not missing anything.

sjcbulldog, 2018-08-25:
It is not really a project from last year.  Last year we used make to build and deploy.  I use Visual Studio Code at work (Cypress Semiconductor)  and when it was announced as the basis for First Development going forward we immediately moved to team to Visual Studio Code and GradleRIO as it stood then.  This was when word first leaked into ChiefDelphi (don't remember the exact date).   In addition, last year was my first year as the lead software mentor for the team (1425).  Last year we used what the software architecture that had been around for some time.  This summer we are working on a much better software architecture for going forward and this is the work you see here.    I say all of this because it is all in a state of transition.

So to be clear on what I know and what I don't know ...

This gradle setup here has been derived from the early GradleRIO download when this direction was first leaked onto Chief Delphi.

Because we are creating a whole new architecture this summer, the students write code outside of robotics meetings.  I created a rough robot simulator to given them something to write against.  It replaces the WPILIB when running locally on the PC and simulates the robot we had from the 2018 year. 
 Our goal is to get the new architecture running on the 2018 robot.  The projects that are for the platform 'x86_64' are cygwin projects and are part of this simulation environment.  The simulator basically depends on a standard C++ environment plus ncurses.  What is here may only work on Windows for now after the recent merge of joystick support which is not cross platform yet.

We generally work using a mix of Linux and Cygwin.  Students and mentors running on Windows assume cygwin and run from this environment.  Visual Studio Code is setup to use cygwin bash as the "Command" shell.  Builds from the command line are run from bash under cygwin.  The debugger is setup locally to use cygwin gdb to debug on the simulator.

This setup works to build code for both the simulator and for the robot.  The robot code works (as well as the code base works) so the robot build is working fine.  However, deploy does not work.  For now we use a deploy script we used with make last year to deploy the robot code.  We have tabled figuring out the gradle deploy problem until we have to the alpha versions of the Visual Studio Code plugin plus GradleRIO running.  That is what I was looking at. 

Note, I know this is obvious, but if you take this and don't run the installToolchain build target, then only the simulator builds which means 18 actionable tasks.  If everything builds, it should be 30 actionable tasks.

[master.zip](https://github.com/wpilibsuite/GradleRIO/files/2320868/master.zip)

If it is easier, you can clone this code from github via 

https://github.com/errorcodexero/newarch.git

Thanks
Sjcbulldog

JacisNonsense, 2018-08-26:
@sjcbulldog this is fixed now. The new version of GradleRIO isn't published yet, but you can get the changes now by cloning GradleRIO, switching to the `backtrack2019` branch, then running `./gradlew publishToMavenLocal -PlocalPublish`, then updating the version in your build.gradle to `2019.0.0-alpha-4-pre1`

closed 2018-08-26 by JacisNonsense

sjcbulldog, 2018-08-26:
This worked perfectly including building all of the cygwin targeted simulation code.  Thank you for the quick response and thanks for the effort with VS Code and Gradle.  This is going to be a much better development environment (in my opinion) than the eclipse based environment.

commit e255024b (2018-08-26): Fix #139