wpilibsuite/GradleRIO #142: Unable to run any gradle commands

Ewpratten, 2018-08-29:
whenever I try to run gradle in a c++ project (without the toolchain installed) i get this:
```
FAILURE: Build failed with an exception.

* Where:
Build file '/media/removable/SD Card/barebonesfrc/build.gradle' line: 3

* What went wrong:
An exception occurred applying plugin request [id: 'edu.wpi.first.GradleRIO', version: '2019.0.0-alpha-3']
> Failed to apply plugin [class 'jaci.gradle.deploy.DeployPlugin']
   > Could not create an instance of type jaci.gradle.deploy.DeployExtension_Decorated.
      > org.gradle.api.tasks.TaskContainer.register(Ljava/lang/String;Lorg/gradle/api/Action;)Lorg/gradle/api/tasks/TaskProvider;

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 5s
```

ThadHouse, 2018-08-29:
Are you using gradle 4.9? Its required for GradleRIO.

Ewpratten, 2018-08-29:
> ./gradlew --version Will tell you what version of the wrapper is
> installed.

Ah. I'm using 4.4.1
That's probably the problem.

ThadHouse, 2018-08-29:
Yup thats definitely the problem.

closed 2018-08-29 by ThadHouse