wpilibsuite/GradleRIO #148: Java jdk error

braydencoyer, 2018-09-10:
> Task :compileJava FAILED
> 
> FAILURE: Build failed with an exception.
> 
> * What went wrong:
> Execution failed for task ':compileJava'.
> > Could not find tools.jar. Please check that C:\Program Files\Java\jre1.8.0_181 contains a valid JDK installation.
> 
> * Try:
> Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
> 
> * Get more help at https://help.gradle.org
> 
> BUILD FAILED in 24s
> 1 actionable task: 1 executed
> The terminal process terminated with exit code: 

this is the error I get whenever we try to build a brand new project and we even changed the java in the settings and preferences. Any help appreciated!

JacisNonsense, 2018-09-10:
Does what it says on the tin, you need to have a JDK installed, not a JRE (JDK is for development, JRE is for running already-built java applications only)

braydencoyer, 2018-09-10:
We have pointed it towards a JDK but it still doesn't change, we have done this both on the settings and preferences.

JacisNonsense, 2018-09-10:
If you're changing them in an IDE, it will have no effect. For gradle to find it, you have to modify your JAVA_HOME environment variable for your system, or delete the JRE

braydencoyer, 2018-09-11:
I just had to delete my JRE's thanks

closed 2018-09-11 by braydencoyer