wpilibsuite/GradleRIO #151: Delegated Dependencies include dependencies for all platforms.

ThadHouse, 2018-09-14:
Working on adding desktop build, I noticed MSVC builds were failing. Looking at the logs, Desktop libraries were being linked into the RoboRIO build (this was not failing), and Rio libraries were being linked into the desktop build (this was failing). 

All that was done was adding `targetPlatform 'desktop'` to a C++ build with the newest GradleRIO.

ThadHouse, 2018-09-14:
Note its not specific to the JSON stuff I'm working on, as its adding wpilib deps as well.

```
-o
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\exe\\frcUserProgram\\roborio\\frcUserProgram
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\objs\\frcUserProgram\\roborio\\frcUserProgramCpp\\biznmyfp3z431mmtf5jlwqjsh\\main.o
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\wpilibc-2018.4.1-windowsx86-64.zip_4046760d02bba87cf419ea8487107d64\\windows\\x86-64\\shared\\wpilibc.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\wpilibc-2018.4.1-windowsx86-64.zip_4046760d02bba87cf419ea8487107d64\\windows\\x86-64\\static\\wpilibc.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\hal-2018.4.1-windowsx86-64.zip_e9cc46a9246ea24265c6e8521bdc15d4\\windows\\x86-64\\shared\\wpiHal.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\hal-2018.4.1-windowsx86-64.zip_e9cc46a9246ea24265c6e8521bdc15d4\\windows\\x86-64\\static\\wpiHal.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\wpiutil-cpp-3.2.0-windowsx86-64.zip_fce6ddee37aa2209a29bc397e1cdbd50\\windows\\x86-64\\shared\\wpiutil.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\wpiutil-cpp-3.2.0-windowsx86-64.zip_fce6ddee37aa2209a29bc397e1cdbd50\\windows\\x86-64\\static\\wpiutil.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\ntcore-cpp-4.1.0-windowsx86-64.zip_4ba95f083c866153120a7c54a35bf646\\windows\\x86-64\\shared\\ntcore.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\ntcore-cpp-4.1.0-windowsx86-64.zip_4ba95f083c866153120a7c54a35bf646\\windows\\x86-64\\static\\ntcore.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\cscore-cpp-1.3.0-windowsx86-64.zip_90a26632bc65f996d0ade5630f1b7323\\windows\\x86-64\\shared\\cscore.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\cscore-cpp-1.3.0-windowsx86-64.zip_90a26632bc65f996d0ade5630f1b7323\\windows\\x86-64\\static\\cscore.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_calib3d320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_core320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_features2d320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_flann320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_highgui320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_imgcodecs320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_imgproc320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_ml320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_objdetect320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_photo320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_shape320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_stitching320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_superres320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_video320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_videoio320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\shared\\opencv_videostab320.lib
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\opencv-cpp-3.2.0-windowsx86-64.zip_6c914e7ab7824a12b5a05c05fb786160\\windows\\x86-64\\static\\opencv320.lib
[... cut ...]
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\ni-libraries-2018.4.1-linuxathena.zip_2c64ecd4b3048d6e908a9a8357e6be0a\\linux\\athena\\shared\\libniriodevenum.so.17.0.0
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\ni-libraries-2018.4.1-linuxathena.zip_2c64ecd4b3048d6e908a9a8357e6be0a\\linux\\athena\\shared\\libniriosession.so.17.0.0
C:\\Users\\thadh\\Documents\\JavaDebugTest\\build\\tmp\\expandedArchives\\ni-libraries-2018.4.1-linuxathena.zip_2c64ecd4b3048d6e908a9a8357e6be0a\\linux\\athena\\shared\\libvisa.so
--sysroot
C:\\frc
-pthread
-rdynamic

```
Heres a log.

JacisNonsense, 2018-09-26:
Fixed in ET 2018.09.27 (commit 821409a)

closed 2018-09-26 by JacisNonsense